stangreg/Ignitron #4: More ESP code if of any use

paulhamsh, 2021-12-03:
This repo was linked by someone on the Spark Amp Mods facebook group - great work!!
There is also a Spark Amp Programming facebook group

My repos have moved on since the python code - I have fully documented the messages to and from the Spark and my latest ESP code can work with the app and the amp together - it is ESP32 code. 
It now works with BLE and Classic bluetooth so works with IOS and Android apps.
Some of it might be of use.

SparkMIDI is the latest code
Spark-Bluetooth-Message-Format is the documentation

stangreg, 2021-12-04:
Thank you very much, it was great fun to work on this, especially when things start working!
I currently don’t have a Facebook account, maybe I will create one to see what is in the group…

I have seen your documentation recently, great stuff! It helped already cleaning the initial messages I am sending when connecting my pedal to the app.
The only thing I am struggling is how I can create a proper license message to enable the Hendrix stuff in the pedal. I have purchased it on iOS, and the current workaround is to first connect the app with the amp, then the pedal.
Have you been able to find the structure of these messages? Of course given that a proper license has been obtained.

All in all very interesting stuff, I will check your repo later to see if there is more to use.

paulhamsh, 2021-12-04:
The license is encrypted I think - it is something like 64 bytes but they change each time they are sent.
I think the only way around it is to connect the app first or keep it connected all the time - my new code does that, connects to app and amp at the same time - but it still needs the app there.

stangreg, 2021-12-04:
I have not done massive testing on that, but when I tried, I just had to connect the app once to enable the license. Afterwards, I could change to a Hendrix preset stored on the pedal and it worked. The app was at that time no longer connected.
I am not connecting both at the same time because the pedal is also acting as a BT keyboard to the mobile to control looper apps or similar. So that would not be an option (at least not simultaneously).
I guess as long as we don't know the encryption and which keys are used the workaround is the only way now.

paulhamsh, 2021-12-04:
Yes I saw you were doing that to control the app. I will study your code to see how that works!

closed 2021-12-22 by stangreg