stangreg/Ignitron #5: Software Building Issues

ServError, 2021-12-27:
Wanted to document the following while trying to build via Arduino IDE:

- First, unrelated to Arduino IDE, the Sloeber hyperlink goes to a malvertising site rather than the plugin page.
- ArduinoJson version requirement listed at 6.8.15. Think you wanted 6.18.5
- Line 11 of SparkDataControl.h has an additional '>' what breaks library resolution.
- Line 12 of SparkPresetBuilder.h spells ArduinoJson.h as ArduinoJSON.h which breaks library resolution.
- There are redefinition errors due to the use of both the standard ESP32 BLE library (via ESP32-BLE-Keyboard) and the nimBLE library. ESP32-BLE-Keyboard can compensate by checking for a define. I added a define for USE_NIMBLE prior to including BleKeyboard.h in both SparkBLEControl.h and SparkDataControl.h.
- SparkDataControl.cpp calls a nonexistent bleKeyboard.start which I replaced with bleKeyboard.begin
- Side note for others: ESP32-BLE-Keyboard needed to be downloaded manually and added via zip, other libraries are in the library manager.

Let me know if you'd like a PR for these changes.

stangreg, 2021-12-27:
Thanks for pointing out these things!
Regarding the items:
1) fixed, I have changed the link to a more appropriate one
2) fixed
3) fixed. Strange that it compiles in the Sloeber IDE
4) Fixed. Moved the include from the .ccp to the .h file
5) Fixed. However, it does not work for me, I have to include that directive in the build parameters via -D parameter.
6) That is a tricky one. I forgot that I have actually patched my local copy of BleKeyboard to start and stop the server. Using begin() will most likely not work as this will try to init the BLE server again (which has already happened). The start and stop methods I implemented look the following:

void BleKeyboard::end(void)
{
	BLEServer *pServer = BLEDevice::getServer();
	for (int i = 0; i < pServer->getConnectedCount(); i++) {
		pServer->disconnect(pServer->getPeerInfo(i).getConnHandle());
	}
	pServer->stopAdvertising();
}

void BleKeyboard::start(void) {
	BLEServer *pServer = BLEDevice::getServer();
	pServer->startAdvertising();

}

Any suggestions if that can be done better?

stangreg, 2021-12-27:
Update: Regarding the start() and stop() I am considering implementing an inheriting class to have everything separated properly.

stangreg, 2021-12-27:
Update: I have now implemented a new class SparkBLEKeyboard which inherits from BleKeyboard. BleKeyboard is back to original, unpatched form.
@ServError I can only build BleKeyboard without including the standard (non-NimBLE) BLE library when using the compile option "-D USE_NIMBLE". Any suggestions how to do it better are welcome.

FrancisDu09, 2021-12-27:
Hello and thank you for the work done. 
Please tell me if I shouldn't post here, but that's to complete the topic.
I am also trying to compile the app but after spending some time trying with Sloeber plugin I gave up and am now trying with Arduino IDE.
After multiple problems, the last version allowed me to move forward but now I get stuck on the following errors:

If someone can help me, thanks !
Francis

sketch\src\SparkBLEKeyboard.cpp.o:(.rodata._ZTV16SparkBLEKeyboard[vtable for SparkBLEKeyboard]+0x20): undefined reference to `BleKeyboard::onConnect(NimBLEServer*)'
sketch\src\SparkBLEKeyboard.cpp.o:(.rodata._ZTV16SparkBLEKeyboard[vtable for SparkBLEKeyboard]+0x24): undefined reference to `BleKeyboard::onDisconnect(NimBLEServer*)'
sketch\src\SparkBLEKeyboard.cpp.o:(.rodata._ZTV16SparkBLEKeyboard[vtable for SparkBLEKeyboard]+0x28): undefined reference to `BleKeyboard::onWrite(NimBLECharacteristic*)'
sketch\src\SparkBLEKeyboard.cpp.o:(.rodata._ZTV16SparkBLEKeyboard[vtable for SparkBLEKeyboard]+0x3c): undefined reference to `non-virtual thunk to BleKeyboard::onConnect(NimBLEServer*)'
sketch\src\SparkBLEKeyboard.cpp.o:(.rodata._ZTV16SparkBLEKeyboard[vtable for SparkBLEKeyboard]+0x44): undefined reference to `non-virtual thunk to BleKeyboard::onDisconnect(NimBLEServer*)'
sketch\src\SparkBLEKeyboard.cpp.o:(.rodata._ZTV16SparkBLEKeyboard[vtable for SparkBLEKeyboard]+0x74): undefined reference to `non-virtual thunk to BleKeyboard::onWrite(NimBLECharacteristic*)'
collect2.exe: error: ld returned 1 exit status
Multiple libraries were found for "WiFi.h"
 Used: C:\Users\francis\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\libraries\WiFi
 Not used: C:\Program Files (x86)\Arduino\libraries\WiFi
 Not used: C:\Users\francis\Documents\Arduino\libraries\WiFi-1.2.7
exit status 1
Error compiling for board ESP32 Wrover Module.

stangreg, 2021-12-27:
@FrancisDu09 Have you included the library "ESP32-BLE-Keyboard" to your project? As mentioned in the original post of this issue, the library needs to be added manually as it is not included in the library manager.

FrancisDu09, 2021-12-27:
Hi and thanks,
I think I'm ok with the library.
For Sloeber I hab multiple error but I just tried with new code and it seems better. I now get an error on lib esp32-hal-i2c.h, I look at that and I tell you.

[image]

ServError, 2021-12-28:
Hi @stangreg , thanks for the quick reply and fixes. It looks like the design of ESP32-BLE-Keyboard does indeed cause the define to not work as expected, but as it turns out this is already a known issue - https://github.com/T-vK/ESP32-BLE-Keyboard/pull/111#issuecomment-954876073

FrancisDu09, 2021-12-28:
Hi @stangreg  and @ServError and thanks.
Finally I managed to compile the program thanks to the last version and the modifications indicated by @ServError to manually modify the BleKeyboard.cpp and BleKeyboard.h.
I manage to send the program to my esp32-wrover but I have errors on startup (please tell me if I have to go through another way to ask these kind of questions).
I am a beginner on this type of material and environment and I admit that everything seems confused to me for the moment.

Serial console : it loops on this at startup.

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
Initializing
Initial button setup
Operation mode: 1
======= Entering APP mode =======
E (878) SPIFFS: mount failed, -10025
E (878) SPIFFS: mount failed, -10025
ERROR while trying to open presets list file
Starting scan
Initialization done.
!!! Restarting !!!
ets Jun  8 2016 00:22:57

stangreg, 2021-12-28:
@FrancisDu09  think you are close now.
What is missing is to put the files in the data folder.
For this, you need to make sure to install the plug-in “ESP32 Sketch Data Upload” to your Arduino IDE. This can be used to upload all data from the data folder for an initial load.
You can just use the files I put there or adjust as per your needs. There is one file per preset and the file PresetList.txt is to be used to sort them accordingly into the banks. If a bank is not filled completely, the program will duplicate the last preset in the list until the bank is full.

Before uploading, make sure to configure the right partition scheme (see the Readme file): 

Partition scheme: Minimal SPIFFS (Large APPS with OTA)

Maybe you want to start with a minimal preset list, then you can add new presets you like via the mobile Spark App.

FrancisDu09, 2021-12-28:
I added the plugin and uploaded the files.
Now I get the following log which also loops.
I only have a development kit with an ESP32-WROVER without any screen or button, it was to see if I could already install the app.
I will order additional material if necessary to advance in testing.

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
Initializing
Initial button setup
Operation mode: 1
======= Entering APP mode =======
Starting scan
Initialization done.
!!! Restarting !!!
ets Jun  8 2016 00:22:57

stangreg, 2021-12-28:
@FrancisDu09 looking at the log output, I think the missing presets were not the reason for the reboots. 
The only place where "!!! Restarting !!!" is logged is when button 2 is long pressed.
So for some reason it seems that the pin configured for button 2 is constantly on, which would cause this effect.
To verify this, you could rebuild after uncommenting the "#define DEBUG" in the file Common.h.
Then after restarting you should see a line like "Button long pressed" before the restart.
If that is the case, you need to reconfigure the Preset 2 button to a different pin. This can be done in the file SparkButtonHandler.h. Please change both values for BUTTON_PRESET2_GPIO and BUTTON_MOD_GPIO.

FrancisDu09, 2021-12-28:
Thanks
Now I've got that kind of log

I NimBLEScan: "Updated advertiser: 4e:77:13:0c:f4:25"
I NimBLEScan: "Updated advertiser: 4e:77:13:0c:f4:25"
I NimBLEScan: "Updated advertiser: e6:3b:d1:a5:c4:fc"
I NimBLEScan: "Updated advertiser: fe:93:c9:9b:27:65"
I NimBLEScan: "Updated advertiser: 51:14:d4:fa:b9:df"
I NimBLEScan: "Updated advertiser: 51:14:d4:fa:b9:df"
I NimBLEScan: "New advertiser: c4:4d:c5:75:fc:07"
I NimBLEScan: "Updated advertiser: eb:a0:6e:e5:f9:14"
I NimBLEScan: "New advertiser: 4a:37:13:aa:8c:38"
I NimBLEScan: "New advertiser: e4:5d:fc:1d:3a:75"
I NimBLEScan: "Updated advertiser: 51:14:d4:fa:b9:df"
I NimBLEScan: "Updated advertiser: 51:14:d4:fa:b9:df"
I NimBLEScan: "Updated advertiser: 4e:77:13:0c:f4:25"
I NimBLEScan: "Updated advertiser: 4e:77:13:0c:f4:25"
I NimBLEScan: "Updated advertiser: e6:3b:d1:a5:c4:fc"

FrancisDu09, 2021-12-28:
I order a display and I tell you.
Thank you very much.
happy holidays
Francis

stangreg, 2021-12-28:
@FrancisDu09 if the restarting log is gone, Ignitron is now looking for the Spark Amp to connect. Do you have the Spark Amp switched on and *not* connected to the Spark App?
Then Ignitron should connect to the Amp. You should then see lines along setting the amp to preset 1 after connecting. 
If the restarting is gone, you can also switch off the DEBUG again for more cleaned up log information.

FrancisDu09, 2021-12-29:
Hello and thank you @stangreg 
I was finally able to get the program to work with a test circuit, no display at the moment.
Thank you very much for the tremendous work.
I had a bug in use, I create a new 'ticket' for that.
Thanks again @stangreg and @ServError.
I'll com back when I got my display.
Francis

Celticpure, 2021-12-29:
@stangreg Thank you a great project. I am having an issue when compiling you code in ArduinoIDE. I get the following error;
[image]

stangreg, 2021-12-29:
@Celticpure Thank you for your feedback.
The error is strange. If you open the file "SparkBLEKeyboard.cpp", you can see that line 2 is part of a comment block. Somehow your compiler does not recognize this as a comment and tries to interpret it. 
Can you check your setup if there is anything related to this?

Celticpure, 2021-12-29:
> @Celticpure Thank you for your feedback. The error is strange. If you open the file "SparkBLEKeyboard.cpp", you can see that line 2 is part of a comment block. Somehow your compiler does not [...]

I have checked the file and there is nothing different from the file in your repository. I might need to update my IDE perhaps? I will try this any report back. Thanks for your help.

stangreg, 2021-12-29:
I could also be that somehow the first line of the file "SparkBLEKeyboard.cpp" in your version got deleted.
Can you see the "/*" in the first line? If not, you might need to add the line (without quotes, of course).

Celticpure, 2021-12-29:
Yes the "/*" was missing. I have adjusted the file and tried to compile again. But i am getting the error below.
[image]

ServError, 2021-12-29:
@Celticpure Please follow the link in my comment above to fix the BLE Keyboard library, that should resolve your issue.

Celticpure, 2021-12-29:
> @Celticpure Please follow the link in my comment above to fix the BLE Keyboard library, that should resolve your issue.

https://github.com/stangreg/Ignitron/issues/5#issuecomment-1001968123

This link?

ServError, 2021-12-29:
> > @Celticpure Please follow the link in my comment above to fix the BLE Keyboard library, that should resolve your issue.
> 
> [#5 (comment)](https://github.com/stangreg/Ignitron/issues/5#issuecomment-1001968123)
> 
> This link?

Correct

stangreg, 2022-01-04:
As there was no further communication here, I assume this is resolved.

closed 2022-01-04 by stangreg