stangreg/Ignitron #6: Crash after changing to the bank 4 and selecting preset 1

FrancisDu09, 2021-12-29:
Hello and thanks for the great work !

I had this crash but I don't know if it's interesting because I can't reproduce it.
During my first tests, I navigated between the banks, the presets and the different functions and I had the following crash without the possibility of reproducing it:

**Reset after changing bank

hanging to preset  4-2...OK!
[W][SPIFFS.cpp:71] begin(): SPIFFS Already Mounted!
[W][SPIFFS.cpp:71] begin(): SPIFFS Already Mounted!
[W][SPIFFS.cpp:71] begin(): SPIFFS Already Mounted!
[W][SPIFFS.cpp:71] begin(): SPIFFS Already Mounted!
Changing to preset  4-1...OK!
CORRUPT HEAP: Bad tail at 0x3ffd9100. Expected 0xbaad5678 got 0x00000002
CORRUPT HEAP: multi_heap.c:432 detected at 0x3ffd9104
abort() was called at PC 0x40099673 on core 1

ELF file SHA256: 0000000000000000

Backtrace: 0x40094fe0:0x3ffcdb00 0x40095259:0x3ffcdb20 0x40099673:0x3ffcdb40 0x40099c30:0x3ffcdb60 0x40082965:0x3ffcdb80 0x40082996:0x3ffcdba0 0x40089b41:0x3ffcdbc0 0x4000beaf:0x3ffcdbe0 0x40095b26:0x3ffcdc00 0x40095d74:0x3ffcdc20 0x40089c02:0x3ffcdc40 0x40089c2d:0x3ffcdc60 0x40089d81:0x3ffcdc90 0x40129287:0x3ffcdcb0 0x40122341:0x3ffcdf70 0x40122231:0x3ffcdfc0 0x40099c7d:0x3ffcdff0 0x40082a02:0x3ffce010 0x40089b4d:0x3ffce030 0x4000bec7:0x3ffce050 0x40194259:0x3ffce070 0x400d40d2:0x3ffce090 0x400d437e:0x3ffce0b0 0x400d65b5:0x3ffce0d0 0x400d6759:0x3ffce2a0 0x400d211b:0x3ffce2c0 0x40106a6c:0x3ffce2e0 0x4009625e:0x3ffce300

Rebooting...
ets Jun  8 2016 00:22:57**

stangreg, 2021-12-29:
So this does not happen every time?
To see where the error could have been you can install another plugin called 'ESP Exception Decoder' in Arduino IDE. After starting it, you can select the file 'Ignitron.elf' from the Release folder and paste the Backtrace information from above into the upper pane. Then you should see the decoded stack in the lower pane. Does that give any useful information?

FrancisDu09, 2021-12-29:
In fact there is a preset file missing : StudioSession.json, error was loading bank 14.

stangreg, 2021-12-29:
I will check to either add it, or remove it from the PresetList file. As it is one of the factory ones, I might choose the former.

stangreg, 2021-12-31:
Fixed. Removed a double entry and added missing StudioSession.json

closed 2021-12-31 by stangreg