stangreg/Ignitron #8: AMP mode connection for android devices

ServError, 2021-12-30:
I'm finding that I can't get the Spark app to detect the Ignitron in AMP mode. The emulated device is visible in the android bluetooth settings but the app quickly fails when searching. I've tested it on a Samsung Galaxy S20+, an S8, and a Lenovo tablet, all with the same result. I believe it was mainly tested on iOS. Has anyone verified functionality in Android?

Celticpure, 2021-12-30:
I am getting this over and over in serial monitor. @ServError I also see the emulated Ignitron device in my Android Bluetooth setting but the Spark app does no see it. 
[image]

stangreg, 2021-12-30:
@ServError have you tried turning me n Debug mode in Common.h? Then you should be able to see more logs during connection.

@Celticpure can you put your backtrace information into the ESP Exception Decoder to see why it is crashing? Btw: your issue is not related to AMP mode. Please open a separate issue, otherwise it will be difficult to track each issue.

Celticpure, 2021-12-30:
> @Celticpure can you put your backtrace information into the ESP Exception Decoder to see why it is crashing? Btw: your issue is not related to AMP mode. Please open a separate issue, otherwise it will be difficult to track each issue.

Noted, if its not related to AMP mode then what is it related to? Just trying to move this along. There appears to be quite a few issues overall with this code that is supposed to work of the bat with ESP32. 

Anyway I have SparkBox that works very reliably.

ServError, 2021-12-30:
> @ServError have you tried turning me n Debug mode in Common.h? Then you should be able to see more logs during connection.

I haven't yet. I will collect logs and do troubleshooting fairly soon. Since I used an uncommon ESP32 board that holds an 18650 battery and has a micro USB pointed to the side, I needed to desolder the usb port to make things fit and haven't yet attached a harness in its place. In the meantime I use an external serial to USB but it's a bit of a pain to hook up.

stangreg, 2021-12-31:
Die to a missing device I have not tested AMP mode with an Android device.
There might be a possibility that Android devices don't connect via Bluetooth LE. I will look into it and see if anything can be done.

ServError, 2021-12-31:
Below is the serial logging on an AMP mode boot. Not seeing any additional messages for attempted connections from the Spark app.

``
ets Jun  8 2016 00:22:57

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8
Initializing
Initial button setup
Operation mode: 2
======= Entering AMP mode =======
Trying to read file list
Starting NimBLE Server
Advertising Started
JSON:
{"PresetNumber": 127, "UUID":"C6CC85D6-A6F5-4302-AFF6-0060D996A3B1",
"Name":"Sweet Child Of Mine" , "Version":"0.7", "Description":"Slash", "Icon":"icon.png", "BPM": 180.0000,
"Pedals": [
{ "Name":"bias.noisegate",  "IsOn": true,   "Parameters":[0.3083,0.4663] },
{ "Name":"LA2AComp",  "IsOn": false,  "Parameters":[0.0000,0.7126,0.7248] },
{ "Name":"DistortionTS9",   "IsOn": true,   "Parameters":[0.4053,0.3247,0.4773] },
{ "Name":"OverDrivenJM45",  "IsOn": true,   "Parameters":[0.7321,0.8549,0.6433,0.3079,0.8200] },
{ "Name":"Tremolo",   "IsOn": false,  "Parameters":[0.4541,0.6999,0.5962] },
{ "Name":"DelayMono",   "IsOn": true,   "Parameters":[0.2645,0.2742,0.2940,0.6000,0.0000] },
{ "Name":"bias.reverb",   "IsOn": true,   "Parameters":[0.2857,0.4084,0.2895,0.3883,0.5821,0.6500,0.2000] } ],
"Filler":"48"
}

0.31
JSON:
{"PresetNumber": 127, "UUID":"C6CC85D6-A6F5-4302-AFF6-0060D996A3B1",
"Name":"Sweet Child Of Mine" , "Version":"0.7", "Description":"Slash", "Icon":"icon.png", "BPM": 180.0000,
"Pedals": [
{ "Name":"bias.noisegate",  "IsOn": true,   "Parameters":[0.3083,0.4663] },
{ "Name":"LA2AComp",  "IsOn": false,  "Parameters":[0.0000,0.7126,0.7248] },
{ "Name":"DistortionTS9",   "IsOn": true,   "Parameters":[0.4053,0.3247,0.4773] },
{ "Name":"OverDrivenJM45",  "IsOn": true,   "Parameters":[0.7321,0.8549,0.6433,0.3079,0.8200] },
{ "Name":"Tremolo",   "IsOn": false,  "Parameters":[0.4541,0.6999,0.5962] },
{ "Name":"DelayMono",   "IsOn": true,   "Parameters":[0.2645,0.2742,0.2940,0.6000,0.0000] },
{ "Name":"bias.reverb",   "IsOn": true,   "Parameters":[0.2857,0.4084,0.2895,0.3883,0.5821,0.6500,0.2000] } ],
"Filler":"48"
}

0.31
Starting WiFI

.......
Connected to __________
IP address: __________
mDNS responder started
Initialization done.
``

I'm also attaching a wireshark trace of the bluetooth activity, wherein I first try to connect to the Ignitron twice unsuccessfully, wait a short while, then try to connect to the real amp.

[btsnoop_hci.log](https://github.com/stangreg/Ignitron/files/7796664/btsnoop_hci.log)

stangreg, 2022-01-01:
My assumption is that Android devices don't connect via BLE, but via BT classic. I have at some point refactored the code so that it used BLE. This was required as I wanted to be able to connect my mobile via BT audio to the Amp while using my pedal, which is only possible via BLE (in APP mode).
So I guess to enable the AMP mode for Android devices, BT classic mode needs to be implemented as well. 
As I don't have much time at the moment, I don't know yet when I will be able to look into this. I also don't have a clear idea on how to make this work at all yet.

ServError, 2022-01-02:
Yes, from my testing this is in fact the case. Rather than query the Android device to see if it supports BLE (which at this point most modern ones do), they seem to have just stuck entirely with working in BT classic.

As far as the audio + pedal setup, I don't think it's a big issue. As long as I power up the amp and connect the pedal first, I can still connect on Android for audio. If we only use BT classic in AMP mode for present configuration, then the behavior should mirror iOS pretty closely. I'm not sure if we can simulcast both classic and  BLE for this, or if it makes more sense to have two sub-modes for AMP mode to decide the connection type.

I can take a stab at it if you'd like.

stangreg, 2022-01-02:
I was also thinking if it is possible or useful to have both types running at the same type. As BT classic is only required for AMP mode which is chosen during startup, it should be quite straight forward to setup the right one during initialization.
Sure, you can have a go. In the best case it is mostly in the init and the other functions can largely be reused.
Let me know if you have questions on certain functions or the call flows.

dpeddi, 2022-08-14:
Hello,

Any progress on this topic? I've seen you have a branch bt-classic. I've did a quick test but i wasn't unable to get it working.  (or i can't understand how to get it to connect to my android phone)

stangreg, 2022-08-14:
Yes, I had created the branch to test something with the classic BT setup. When I did this, I still had the logic of fixed messages sent for initialization which was not very robust. On the main branch I have recently changed this to be more dynamic (For iOS), so I might pick this up at a later point again. There is also a memory challenge with this as it is getting quite full.

stangreg, 2022-08-17:
Update: The code in the branch **20220817_btclassic_2** does work with Android devices in AMP mode, and ONLY with Android. I will at a later point check for a solution to run both BT BLE and BT Classic in parallel (if possible). If anyone has an idea how to run both stacks in parallel, please let me know.

@dpeddi @ServError @Celticpure

dpeddi, 2022-08-18:
According to following sites it should be possible.

https://github.com/espressif/arduino-esp32/issues/1150

stangreg, 2022-09-01:
As an general update:
I have tried getting to work to be able to switch between BLE (iOS) and BT Serial (Android).
I got only so far that when initializing with BT Serial, I can connect. Then I can switch to BLE and the connection with iOS is working. However, when switching back to BT Serial, it crashes with what I think is memory corruption.
I could not find a way to fix this, maybe something is wrong with the libraries used. 

What I also tested in my code is to not actually switch, but to switch off and on BT Serial, which worked well. Switching BLE of and on worked as well. Only switching from BLE to Serial leads to the crash.

If someone wants to have a look at the code, it is in the branch **20220822_bt_cleanup**

I am about to give up.

ServError, 2022-10-03:
I don't know how I missed the ping in August, but I just noticed it and loaded your branch, and it works wonderfully! Thanks so much for taking the time for it.

I'm sorry that the combined build hasn't worked out. If I get a chance, I'll take a look at the cleanup branch if I get a chance (not that things panned out so well last time I said that). Thanks again @stangreg!

closed 2023-02-25 by stangreg