stangreg/Ignitron #9: Not functioning after successful programming

samiam21, 2022-03-25:
**Describe the bug**
I have successfully programmed the board using Sloeber but none of the LEDs nor the display are working.  I have connected to the serial monitor and do not see any data.  Pressing the buttons does not cause anything to happen either through the serial monitor nor on the LEDs/display.  I have 9v power connected and have tried with the USB both connected and disconnected after programming.

**To Reproduce**
Steps to reproduce the behavior:
1. Load the Ignitron project into sloeber
2. Hold the "boot" button on the ESP32 and program the board
3. After successful program, release the button
4. Power cycle the device
5. Connect to the serial monitor

**Expected behavior**
The display should display something, clicking on the buttons should turn LEDs on and off, and something should be displaying over the serial monitor.

**Screenshots**
Images of the successful programming:
[image]
[image]

**Desktop (please complete the following information):**
 - OS: Windows 10
 - IDE: Sloeber 4.4

stangreg, 2022-03-26:
The very first thing which happens when starting the Ignitron is a Serial message "Initializing" and a logo appearing in the display. If that is not seen, I assume there must be something wrong with the setup or the hardware.
Different options what you can do to get more insight:
- Check if you have really connected to the Serial port
- Try to install a basic example to the board using the standard Arduino IDE and installing one of the example projects. Then you can see if the general setup works.
- Read carefully through the Ignitron build instructions if something was missed when setting up the Sloeber environment.
- Enable Debug logging in Ignitron by removing the Comment in the "Common.h" file and rebuild

Please try the above and let me know if in any of the options above you can see something in the Serial log.

samiam21, 2022-03-26:
Thank you for your response @stangreg!

I loaded the Arduino blinky example and updated it to write out the serial port with each blink.  I programmed that and was able to see the serial writes coming through on my COM port.

I verified that Sloeber is set up according to the instructions and that all of the libraries are the same version as listed.  I enabled debug logging and an receiving this error for SPIFFS:

`rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8
Initializing
Initial button setup
Operation mode: 1
======= Entering APP mode =======
Trying to read file list
E (11) SPIFFS: mount failed, -10025
E (12) SPIFFS: mount failed, -10025
ERROR while trying to open presets list file
Starting scan
Initialization done.`

stangreg, 2022-03-26:
I would suspect that you have not uploaded the data folder with the presets to the Ignitron.
You need to install the data uploader plug-in to the Arduino IDE and upload the data folder using the Arduino IDE. Please make sure to choose the right partitioning before as per documentation.

samiam21, 2022-03-26:
I uploaded the data file and reprogrammed the board.  I am still not getting anything on the screen but see this in the debug console:

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8
Initializing
Initial button setup
Operation mode: 1
======= Entering APP mode =======
Trying to read file list
Last bank not full, filling with last preset to get bank complete
Last bank not full, filling with last preset to get bank complete
Starting scan
Initialization done.

Thank you again, so much, for the help!

stangreg, 2022-03-27:
The log file is exactly as expected.
Regarding the screen, I would assume that there is something wrong with the wiring or the hardware. 
Have you tried switching on the Spark amp to see if the board connects to it? You need to wait for a moment, then you should see more log files about the connection.
If that looks fine, I would assume that everything is fine with the software.

stangreg, 2022-04-03:
@samiam21 Did you get it to work?

samiam21, 2022-04-04:
@stangreg the display isn't powering on still so my buddy (this is his project, I'm just helping him to load the software on it) is going to get a new display and check all the wiring.  Not sure when he will be able to do that so we can definitely close out this ticket as it doesn't seem to be any software issue.  I really appreciate all of your help!!