stangreg/Ignitron #14: Ignitron crash in app mode changing banks

dpeddi, 2022-08-15:
**Describe the bug**
Ignitron crash in app mode while changing banks

I've tried with platformio and with sloeber + arduino and i get the same issue.
I've uploaded the sketch and uploaded the spiffs. 

```
rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5856
entry 0x400806a8
Initializing
Initial button setup
Operation mode: 1
                 ======= Entering APP mode =======
Last bank not full, filling with last preset to get bank complete
Last bank not full, filling with last preset to get bank complete
Last bank not full, filling with last preset to get bank complete
Starting scan
Initialization done.
SparkDataControl not setup yet, ignoring button press.
Found Spark, connecting.
Scan ended.
Connected to: f7:eb:ed:x.x.x
Subscribing to service notifications of FFC2
                                            Notifications turned on
Done with this device.
BLE connection to Spark established.
Changing to HW preset 1
                       Changing to HW preset 1
, returning empty preset.Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x4019385f  PS      : 0x00060530  A0      : 0x800d67d0  A1      : 0x3ffcd8e0
A2      : 0x3ffcda84  A3      : 0x00000000  A4      : 0x3f41c26b  A5      : 0x3ffd40b0
A6      : 0x00000010  A7      : 0x00000001  A8      : 0x3ffcda8c  A9      : 0x3ffcd8b0
A10     : 0x3ffcda5c  A11     : 0x3f41c26a  A12     : 0x3f41c26b  A13     : 0x0000ff00
A14     : 0x00ff0000  A15     : 0xff000000  SAR     : 0x00000008  EXCCAUSE: 0x0000001c
EXCVADDR: 0x00000000  LBEG    : 0x4000c349  LEND    : 0x4000c36b  LCOUNT  : 0xffffffff

ELF file SHA256: 0000000000000000

Backtrace: 0x4019385f:0x3ffcd8e0 0x400d67cd:0x3ffcd900 0x400d6a47:0x3ffcdad0 0x400d1cfb:0x3ffcdaf0 0x40103140:0x3ffcdb10 0x40091082:0x3ffcdb30

Rebooting...
```

The log in AMP mode is more speaking... it seems that the json are unreadable.
```
                 ======= Entering AMP mode =======
Last bank not full, filling with last preset to get bank complete
Last bank not full, filling with last preset to get bank complete
Last bank not full, filling with last preset to get bank complete
Starting NimBLE Server
Advertising Started
, returning empty preset.Starting WiFIpening file /SweetChildOfMine.json

........
Connected to OpenWrt-WDS
IP address: 192.168.179.140
mDNS responder started
Initialization done.
, returning empty preset./SweetChildOfMine.json
```

I've downloaded the spiffs from the esp with download_fs.py from https://github.com/maxgerhardt/pio-esp32-esp8266-filesystem-downloader and the json are good.

stangreg, 2022-08-15:
What is strange here is that in APP mode it seems to crash after changing to HW preset 1, which is not even read from the file system. Changing to HW presets are special commands directly to the amp.

Are you by any chance using a Spark Mini? I have only tested the code with the Spark 40 amp, there might be some differences in operating the mini when it comes to the HW presets.

In AMP mode, it seems that there was an error with opening the "SweetChildOfMine.json" preset. It is hard to tell from remote why it could not read the file.

have you tried copying the backtrace information into the Exception Decoder (add-on for the Arduino IDE) to see what the exception stack looks like?

stangreg, 2022-08-15:
Some things which would help to know the issue better:
- what is shown in the display after start until the issue occurred?
- have you pressed any buttons before the issue occurred? If so, what was shown in the display before pressing the button?
- Please describe all steps you are doing after powering on the pedal and the amp.

dpeddi, 2022-08-15:
Hello,

here is the backtrace...
```
Decoding stack results
0x40193983: std::__cxx11::basic_string  , std::allocator  >::basic_string(std::__cxx11::basic_string  , std::allocator  > const&) at /builds/idf/crosstool-NG/.build/xtensa-esp32-elf/build/build-cc-gcc-final/xtensa-esp32-elf/libstdc++-v3/include/bits/basic_string.h line 135
0x400d67f9: SparkDisplayControl::showFX_SecondaryName() at C:\...\Arduino\Ignitron\src/SparkTypes.h line 38
0x400d6a73: SparkDisplayControl::update(bool) at C:\...\Arduino\Ignitron\src/SparkDisplayControl.cpp line 274
0x400d1d27: loop() at C:\...\Arduino\Ignitron/Ignitron.ino line 102
0x4010285c: loopTask(void*) at C:\...\AppData\Local\Arduino15\packages\esp32\hardware\esp32\1.0.6\cores\esp32\main.cpp line 23
0x40091082: vPortTaskWrapper at /home/runner/work/esp32-arduino-lib-builder/esp32-arduino-lib-builder/esp-idf/components/freertos/port.c line 143
```
```
	if(fileSystem.openFromFile(&presetFilename[0], presetJsonString)){
		retPreset = getPresetFromJson(&presetJsonString[0]);
		return retPreset;
	}
	else{
		Serial.printf("Error while opening file %s, returning empty preset.", presetFilename.c_str());
		return retPreset; --> here retPreset is an empty structure.. 
	}
```
I think we could investigate and fixed but could be related to the empty data.

stangreg, 2022-08-15:
Thanks.
Another thing which might help: can you please go to Common.h and remove the comment from the line #define DEBUG, then rebuild? This should give a more detailed log.

And, are you using Spark Mini or the big one Spark 40?

dpeddi, 2022-08-15:
I've the spark 40

```
	std::string presetFilename = "/"+presetBanksNames[bank-1][pre-1];
	std::string presetJsonString;

	Serial.printf("****************************************************\n");

	std::string scom = "/SweetChildOfMine.json";

	if(fileSystem.openFromFile(scom.c_str(), presetJsonString)) {
		Serial.println("Eddi OK");
	} else {
		Serial.println("Eddi KO");
	}

	Serial.printf("****************************************************\n");

	Serial.printf("-->%s<--",&presetFilename[0]);
	Serial.printf("****************************************************\n");
	sleep(1);

	if(fileSystem.openFromFile(&presetFilename[0], presetJsonString)){
		retPreset = getPresetFromJson(&presetJsonString[0]);
		return retPreset;
	}
	else{
		Serial.printf("Error while opening file %s, returning empty preset.", presetFilename.c_str());
		return retPreset;
	}
}
```

```
Button pressed: 18
****************************************************
Eddi OK
****************************************************
-->/SweetChildOfMine.json
<--****************************************************
Error while opening file /SweetChildOfMine.json
, returning empty preset.
```

I seems it could depend on the newline... you develop under windows or linux?

I'm using windows and i might have in the txt the \r also...

dpeddi, 2022-08-15:
I think it could be fixed with:
	while (std::getline(stream, line)) {
		**line.erase(std::remove(line.begin(), line.end(), '\r' ), line.end());
		line.erase(std::remove(line.begin(), line.end(), '\n' ), line.end());**

or you could just document to double check that the file must be with unix newline.
Regards

stangreg, 2022-08-15:
This might be the case, I am developing on Mac OS. You could try to apply the fix at your end and if it is working for you, I can test if it still works on my end.

dpeddi, 2022-08-15:
It worked for me,
Maybe it could be wrote better...

stangreg, 2022-08-15:
Great to hear that it works for you. I am currently on a business trip, once I am back I will see what can be done best. Thanks for your support!

dpeddi, 2022-08-15:
You are welcome!

closed 2022-09-01 by stangreg