stangreg/Ignitron #17: Firmware

Storgio, 2022-12-23:
Hi, I’m completely new on this topic but I would love to have a working ignitron for my spark. 
first of all thanks for the great project then… would be possible to have a firmware to upload without the needing of compiling it? I have no idea where to start from..

stangreg, 2023-01-02:
Hi, I could upload the files, however, I am not sure what files you would require. Also, given that you have no idea where to start from, I would think it would still be difficult for you to get it to work easier as if you would compile it.
I would recommend to follow the build instructions. The new Arduino IDE is very useful and if you add all the required libraries as documented, it should work.
So please read the documentation and try.

stangreg, 2023-01-04:
Hi,

1. I think the FS library is part of the standard board installation. So you don't need to include anything, it should work out of the box
2. you have to find the file platform.txt in the correct Arduino folder (for me it was in 'packages/esp32/hardware/esp32/2.0.6/platform.txt' in the Arduino folder). Then you have to find the line starting with "build.extra_flags.esp32" and add " -DUSE_NIMBLE" at the end.
3. This is likely an error due to the missing USE_NIMBLE flag (see 2.)

Please try again and let me know if it worked.

Storgio, 2023-01-04:
Ya man, it worked!
I managed to compile, upload and upload the data folder.
The only error i got now is: "ERROR while to trying to open preset list file"
Seems like it's missing a sort of "index" file for the presets in the data folder, right?

The hardware is not complete and I will try to connect it to Spark right now...

Anyway: once again THANKS for your project, your help, your support and your kindness.

stangreg, 2023-01-04:
OK, we are one step closer. 
The error seems to indicate that the data folder was not uploaded correctly. The index file in the data folder is called "PresetList.txt" and should be uploaded together with the presets.

As you described above, you are using Arduino 2.0 IDE. This is ok for compiling, but this version unfortunately does not support external tool support. As described in the documentation, the "ESP32 Sketch Data Upload" tool is required to upload the data. In order to use this, you need to install the version 1.8.19 of the Arduino IDE. There you can install the Data Upload plugin and use it to upload the data. 
I hope Arduino will add tools support soon to fully enable IDE 2.0.

Let me know if that worked.

Storgio, 2023-01-04:
I setted up Arduino 1.8 to upload the data folder but the preset list file is not in the data folder..

stangreg, 2023-01-04:
When you check here on GitHub, there is definitely a folder called "data" on top level. Please check if that folder is also present and filled in your local copy. There should be a file called "PresetList.txt" and a number of .json files containing the presets.
To upload the data folder, make sure to open the Ignitron.ino project in the Arduino IDE (1.8) and Choose "ESP Sketch Data Upload" in the "Tools" menu. This should then do the trick. Make sure to choose the correct partition scheme in the Tools folder (Minimal SPIFFS (Large APPS with OTA))

Storgio, 2023-01-05:
DONE!
I manage to upload the folder and I have no more errors.
The only problem was that I had to reduce the folder size by deleting some files because otherwise I mhad a "file system full" error, I don't know why.

Is it possible that I added multiple times the same files? How do I check the data I upped?
Thanks again

stangreg, 2023-02-25:
I don't think that could upload data multiple times as it will always write to the same memory location.
Maybe you have less memory on your board. If it works now that is great to hear. I will close this as the original issue is resolved.

closed 2023-02-25 by stangreg

Storgio, 2023-03-01:
That's what I get...SUUUPER LONG!

Button long pressed: 26
!!! Restarting !!!
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13220
ho 0 tail 12 room 4
load:0x40080400,len:3028
entry 0x400805e4
Initializing
Initial button setup
Operation mode: 2
======= Entering AMP mode =======
Trying to read file list
Starting NimBLE Server
Advertising Started
JSON:
{"PresetNumber": 127, "UUID": "D4351149-185B-4959-B831-B778C62B35BA", 
"Name": "Sultans Of Swing / Knopfler", "Version": "0.7", "Description": "1-Clean", "Icon": "icon.png", "BPM": 120.0000, 
"Pedals": [
{"Name": "bias.noisegate", "IsOn": true, "Parameters":[0.1201, 0.3314, 1.0000]}, 
{"Name": "Compressor", "IsOn": false, "Parameters":[0.3335, 0.9991]}, 
{"Name": "Booster", "IsOn": true, "Parameters":[0.3445]}, 
{"Name": "OverDrivenLuxVerb", "IsOn": true, "Parameters":[0.2969, 0.2304, 0.2840, 0.3829, 0.8264]}, 
{"Name": "ChorusAnalog", "IsOn": false, "Parameters":[0.0939, 0.2797, 0.7458, 0.5678]}, 
{"Name": "DelayRe201", "IsOn": false, "Parameters":[0.1913, 0.5975, 0.7797, 0.0000, 1.0000]}, 
{"Name": "bias.reverb", "IsOn": false, "Parameters":[0.4810, 0.2030, 0.3152, 0.3607, 0.6429, 0.3393, 0.4000, 0.0000]}],
"Filler": "56"
}
JSON:
{"PresetNumber": 127, "UUID": "D4351149-185B-4959-B831-B778C62B35BA", 
"Name": "Sultans Of Swing / Knopfler", "Version": "0.7", "Description": "1-Clean", "Icon": "icon.png", "BPM": 120.0000, 
"Pedals": [
{"Name": "bias.noisegate", "IsOn": true, "Parameters":[0.1201, 0.3314, 1.0000]}, 
{"Name": "Compressor", "IsOn": false, "Parameters":[0.3335, 0.9991]}, 
{"Name": "Booster", "IsOn": true, "Parameters":[0.3445]}, 
{"Name": "OverDrivenLuxVerb", "IsOn": true, "Parameters":[0.2969, 0.2304, 0.2840, 0.3829, 0.8264]}, 
{"Name": "ChorusAnalog", "IsOn": false, "Parameters":[0.0939, 0.2797, 0.7458, 0.5678]}, 
{"Name": "DelayRe201", "IsOn": false, "Parameters":[0.1913, 0.5975, 0.7797, 0.0000, 1.0000]}, 
{"Name": "bias.reverb", "IsOn": false, "Parameters":[0.4810, 0.2030, 0.3152, 0.3607, 0.6429, 0.3393, 0.4000, 0.0000]}],
"Filler": "56"
}
Initialization done.
Button pressed: 25
JSON:
{"PresetNumber": 127, "UUID": "D4351149-185B-4959-B831-B778C62B35BA", 
"Name": "Sultans Of Swing / Knopfler", "Version": "0.7", "Description": "1-Clean", "Icon": "icon.png", "BPM": 120.0000, 
"Pedals": [
{"Name": "bias.noisegate", "IsOn": true, "Parameters":[0.1201, 0.3314, 1.0000]}, 
{"Name": "Compressor", "IsOn": false, "Parameters":[0.3335, 0.9991]}, 
{"Name": "Booster", "IsOn": true, "Parameters":[0.3445]}, 
{"Name": "OverDrivenLuxVerb", "IsOn": true, "Parameters":[0.2969, 0.2304, 0.2840, 0.3829, 0.8264]}, 
{"Name": "ChorusAnalog", "IsOn": false, "Parameters":[0.0939, 0.2797, 0.7458, 0.5678]}, 
{"Name": "DelayRe201", "IsOn": false, "Parameters":[0.1913, 0.5975, 0.7797, 0.0000, 1.0000]}, 
{"Name": "bias.reverb", "IsOn": false, "Parameters":[0.4810, 0.2030, 0.3152, 0.3607, 0.6429, 0.3393, 0.4000, 0.0000]}],
"Filler": "56"
}
Button pressed: 26
JSON:
{"PresetNumber": 127, "UUID": "A96E8DD2-4C7A-4BC8-8195-1FEECA4D34E6", 
"Name": "Hendrix ", "Version": "0.7", "Description": "Purple Haze", "Icon": "icon.png", "BPM": 120.0000, 
"Pedals": [
{"Name": "bias.noisegate", "IsOn": true, "Parameters":[0.8193, 1.0000]}, 
{"Name": "LA2AComp", "IsOn": false, "Parameters":[0.0000, 0.7126, 0.7248]}, 
[... 404 lines cut ...]
01FE000041FF6A0000000000000000003C187E2803114A3F191B191A04114A0000F7F0010F020301200F0C1900002B62006961732E72657658657262431700115D4A3E2B605F0111F7F0010F500301480F0D194A3F24500C1402114A3F05532C2603114A3E10062D0E04
Sending data:
01FE000041FF33000000000000000000114A3E7062F7F0010F610301600F0E104E05114A623E69374C06114A1C3F19191A2AF7
0xffc1: onWrite()
Received data:
01FE000053FE3C000000000000000000F0011003020100000300000000000000000000000000000000000000000000000000000000000000000000F7
Reading request from Amp
Found request for current preset
Sending data:
01FE000041FF6A000000000000000000F00110630301200F001901005924004139364538444400322D344337412D00344243382D3831F7F00110780301000F011939352D3100464545434134441033344536284865406E647269782023F7F00110000301400F0219302E
Sending data:
01FE000041FF6A000000000000000000372B00507572706C65201048617A65286963406F6E2E706E674AF7F00110350301100F03194270000003172E626961732E006E6F69736567616C7465431200114AF7F00110770301600F04193F513D251601114A3F0000000128
Sending data:
01FE000041FF6A0000000000000000004C413241436F6C6D70421300114AF7F00110320301000F0519000000004601114A3F366C742602114A3F390C7E612446757A7A4312F7F001102F0301300F061900114A3F301E560401114A3F08286C222A4F7261006E67654144
Sending data:
01FE000041FF6A0000000000000000003330F7F001103D0301580F071943150011594A3F19191A0111414A3F3333330211414A3F08240B0311F7F001103A0301080F08194A3F6E5C0D1204114A3F414102202C43686F72750073416E616C6F67F7F00110430301580F09
Sending data:
01FE000041FF6A0000000000000000001943140011594A3E2D2B1F0111594A3E4C4C4D0211414A3F3333330311F7F001104A0301480F0A194A3F4C4C034D2944656C6179304D6F6E6F4215000B114A3E437B4A01F7F001104A0301580F0B19114A3E4B1B6D7A02114A3F
Sending data:
01FE000041FF6A0000000000000000003C187E2803114A3F191B191A04114A0000F7F00110020301200F0C1900002B62006961732E72657658657262431700115D4A3E2B605F0111F7F00110500301480F0D194A3F24500C1402114A3F05532C2603114A3E10062D0E04
Sending data:
01FE000041FF33000000000000000000114A3E7062F7F00110610301600F0E104E05114A623E69374C06114A1C3F19191A2AF7
Address: 6f:21:50:27:fb:4f Unsubscribed to 0xffc2
Client disconnected
Start advertising