stangreg/Ignitron #18: Rebooting in a Loop

DeadLED66, 2023-02-20:
Hi,

I've worked through a few of the issues that others have had, and I've gotten the firmware uploaded to the ESP-Wroom 32.
Now it's in a loop of constantly restarting.  I've tried redefining the preset button 2 as per one of the other issues. The serial out log is:
`Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0018,len:4
load:0x3fff001c,len:1216
ho 0 tail 12 room 4
load:0x40078000,len:10944
load:0x40080400,len:6388
entry 0x400806b4
E (16) psram: PSRAM ID read error: 0xffffffff
Initializing
Initial button setup
Operation mode: 1
======= Entering APP mode =======
Trying to read file list
Last bank not full, filling with last preset to get bank complete
Last bank not full, filling with last preset to get bank complete
Last bank not full, filling with last preset to get bank complete
Guru Meditation Error: Core  1 panic'ed (LoadProhibited). Exception was unhandled.
Core 1 register dump:
PC      : 0x401d5dff  PS      : 0x00060530  A0      : 0x800d35f3  A1      : 0x3ffca970  
A2      : 0x00000000  A3      : 0x3ffbf0dc  A4      : 0x3ffc69a0  A5      : 0x00000000  
A6      : 0x00000013  A7      : 0x4501fba8  A8      : 0x3ffc5090  A9      : 0x3ffca980  
A10     : 0x00000001  A11     : 0x000006f5  A12     : 0x00000001  A13     : 0x00000001  
A14     : 0x00060520  A15     : 0x00000000  SAR     : 0x00000008  EXCCAUSE: 0x0000001c  
EXCVADDR: 0x00000014  LBEG    : 0x40093ae1  LEND    : 0x40093af1  LCOUNT  : 0xffffffff  

ELF file SHA256: 0000000000000000

Backtrace: 0x401d5dff:0x3ffca970 0x400d35f0:0x3ffca990 0x400d5e2a:0x3ffca9e0 0x400d21ee:0x3ffcacd0 0x4010760a:0x3ffcad10 0x400959a2:0x3ffcad30

Rebooting...`

And so on...

Any help would be appreciated.
Thanks

ServError, 2023-02-20:
What are you using to build the firmware (presumably, which arduino IDE version). Have you tried doing the initial flash with the Erase All Flash option enabled? Did you upload the contents of the data folder per the final paragraph of the Software page?

stangreg, 2023-02-20:
Thanks for your issue report.
Do you have the possibility to decode the backtrace? There are tools available to decode the stack trace. Note that the one for Arduino IDE is only compatible with pre-2.0 versions. 
This could help with the analysis.
How much memory does your board have?
Just a short in the dark, have you tried removing the majority of the JSON files in the data folder and adjusting the PresetList.txt file accordingly? 
Please remember to re-upload the file system afterwards.

stangreg, 2023-02-20:
Ja m Addition, I just saw the following line during boot:

E (16) psram: PSRAM ID read error: 0xffffffff

Not sure if that is related to the issue, but it seems to be an error during initialization, before the code kicks in.

DeadLED66, 2023-02-20:
Hi, I'm using Arduino 1.8 to build, I've also done the upload of the data folder.  I did not try the Erase all option though.  I'll try that and report back.

DeadLED66, 2023-02-20:
Screen shot of exception:
[image]

DeadLED66, 2023-02-25:
Update on this... I switched to using PlatformIO instead of Arduino IDE and this problem has gone away.  Itmust be using a different version of the compiler or something.

closed 2023-02-25 by DeadLED66

stangreg, 2023-02-25:
Nice to hear that it works now!