stangreg/Ignitron #19: iOS connection

Storgio, 2023-02-20:
Hi,
I finished the project and it’s simply amazing, thanks for the great job!

I have a problem uploading presets with iOS because it doesn’t connect: it finds the device but it can’t connect, with Android works fine (I’m using the right Bluetooth settings).
Could you help me?

It would be great to add a couple more switches for looper settings and maybe a tap tempo functions… do you think it would be possible?

stangreg, 2023-02-23:
Regarding the iOS connection, let me give you some points to check, maybe one of them will sort it out:
- If you are building from the main branch, BLE is the standard mode of connecting, which is required for iOS
- Make sure to put your pedal into AMP mode. This can be done by pressing and holding buttons "Preset 1" and "Preset 2" until it restarts. You can verify you are in the right mode if you see a preset in the display and you can browse through them with the bank/preset buttons without being connected to the Spark.
- Turn off the Spark 40 to avoid the app connecting to it or interfering
- In the Spark app, go to the menu (burger menu on the lower right), then swipe the upper image to the left until you see the "Add device" button. Press this button
- The app should now connect to the Ignitron, you can see that if the app says connected and the display shows a bluetooth icon.
- If that worked, you can try sending a preset to the Ignitron by selecting a stored preset or a preset in ToneCloud. The display should show the selected preset name.

Please update here in case this all does not work. 
You can also try enabling debug log (by uncommenting the commented line in "Common.h") and posting the output here.

stangreg, 2023-02-23:
Regarding the additional looper features, I am trying to keep the complexity low, I already feel that the buttons are a bit overloaded. Also, I need to find time to implement this as this is quite time consuming.
Anyway, thanks for the good ideas, I will keep them in mind when I extend functionality.

Storgio, 2023-02-27:
Hi Stan,
when I try to connect the bluetooth logo turns on, I got a Spark BLE connection active in my bluetooth manager on IOS but the Spark app always says "connection failed" and the upload doesnt work.
No new amp is added in the app.
I dunno what happen...

stangreg, 2023-02-28:
Can you send the output of the Serial monitor?
To get more detailed output, you can remove the comment in the file Common.h to enable DEBUG logging and rebuild.

Please try to send both versions, with and without DEBUG.

Storgio, 2023-03-01:
Hi Stan,
here the serial monitor without debug

!!! Restarting !!!
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13104
load:0x40080400,len:3036
entry 0x400805e4
Initializing
Initial button setup
Operation mode: 2
======= Entering AMP mode =======
Starting NimBLE Server
Advertising Started
Initialization done.
Multi-connect support: start advertising
Address: 62:3b:e9:7b:e2:8d Subscribed to notfications for 0xffc2
Address: 62:3b:e9:7b:e2:8d Unsubscribed to 0xffc2
Client disconnected
Start advertising
Multi-connect support: start advertising
lld_pdu_get_tx_flush_nb HCI packet count mismatch (0, 1)
Client disconnected
Start advertising
Multi-connect support: start advertising
Address: 62:3b:e9:7b:e2:8d Subscribed to notfications for 0xffc2
Address: 62:3b:e9:7b:e2:8d Unsubscribed to 0xffc2
Client disconnected
Start advertising
Multi-connect support: start advertising
lld_pdu_get_tx_flush_nb HCI packet count mismatch (0, 1)
Client disconnected
Start advertising
Multi-connect support: start advertising
Address: 62:3b:e9:7b:e2:8d Subscribed to notfications for 0xffc2
Address: 62:3b:e9:7b:e2:8d Unsubscribed to 0xffc2
Client disconnected
Start advertising

The bluetooth logo turns on, in the phone device manager i can see a connection to Spark BLE but then gets disconnected.

stangreg, 2023-03-01:
Thanks, can you try to do the same with DEBUG on in Common.h?

Storgio, 2023-03-01:
That's the log with debug

Button long pressed: 26
!!! Restarting !!!
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:1
load:0x3fff0030,len:1184
load:0x40078000,len:13220
ho 0 tail 12 room 4
load:0x40080400,len:3028
entry 0x400805e4
Initializing
Initial button setup
Operation mode: 2
======= Entering AMP mode =======
Trying to read file list
Starting NimBLE Server
Advertising Started
JSON:
{"PresetNumber": 127, "UUID": "D4351149-185B-4959-B831-B778C62B35BA", 
"Name": "Sultans Of Swing / Knopfler", "Version": "0.7", "Description": "1-Clean", "Icon": "icon.png", "BPM": 120.0000, 
"Pedals": [
{"Name": "bias.noisegate", "IsOn": true, "Parameters":[0.1201, 0.3314, 1.0000]}, 
{"Name": "Compressor", "IsOn": false, "Parameters":[0.3335, 0.9991]}, 
{"Name": "Booster", "IsOn": true, "Parameters":[0.3445]}, 
{"Name": "OverDrivenLuxVerb", "IsOn": true, "Parameters":[0.2969, 0.2304, 0.2840, 0.3829, 0.8264]}, 
{"Name": "ChorusAnalog", "IsOn": false, "Parameters":[0.0939, 0.2797, 0.7458, 0.5678]}, 
{"Name": "DelayRe201", "IsOn": false, "Parameters":[0.1913, 0.5975, 0.7797, 0.0000, 1.0000]}, 
{"Name": "bias.reverb", "IsOn": false, "Parameters":[0.4810, 0.2030, 0.3152, 0.3607, 0.6429, 0.3393, 0.4000, 0.0000]}],
"Filler": "56"
}
JSON:
{"PresetNumber": 127, "UUID": "D4351149-185B-4959-B831-B778C62B35BA", 
"Name": "Sultans Of Swing / Knopfler", "Version": "0.7", "Description": "1-Clean", "Icon": "icon.png", "BPM": 120.0000, 
"Pedals": [
{"Name": "bias.noisegate", "IsOn": true, "Parameters":[0.1201, 0.3314, 1.0000]}, 
{"Name": "Compressor", "IsOn": false, "Parameters":[0.3335, 0.9991]}, 
{"Name": "Booster", "IsOn": true, "Parameters":[0.3445]}, 
{"Name": "OverDrivenLuxVerb", "IsOn": true, "Parameters":[0.2969, 0.2304, 0.2840, 0.3829, 0.8264]}, 
{"Name": "ChorusAnalog", "IsOn": false, "Parameters":[0.0939, 0.2797, 0.7458, 0.5678]}, 
{"Name": "DelayRe201", "IsOn": false, "Parameters":[0.1913, 0.5975, 0.7797, 0.0000, 1.0000]}, 
{"Name": "bias.reverb", "IsOn": false, "Parameters":[0.4810, 0.2030, 0.3152, 0.3607, 0.6429, 0.3393, 0.4000, 0.0000]}],
"Filler": "56"
}
Initialization done.
Button pressed: 25
JSON:
{"PresetNumber": 127, "UUID": "D4351149-185B-4959-B831-B778C62B35BA", 
"Name": "Sultans Of Swing / Knopfler", "Version": "0.7", "Description": "1-Clean", "Icon": "icon.png", "BPM": 120.0000, 
"Pedals": [
{"Name": "bias.noisegate", "IsOn": true, "Parameters":[0.1201, 0.3314, 1.0000]}, 
{"Name": "Compressor", "IsOn": false, "Parameters":[0.3335, 0.9991]}, 
{"Name": "Booster", "IsOn": true, "Parameters":[0.3445]}, 
{"Name": "OverDrivenLuxVerb", "IsOn": true, "Parameters":[0.2969, 0.2304, 0.2840, 0.3829, 0.8264]}, 
{"Name": "ChorusAnalog", "IsOn": false, "Parameters":[0.0939, 0.2797, 0.7458, 0.5678]}, 
{"Name": "DelayRe201", "IsOn": false, "Parameters":[0.1913, 0.5975, 0.7797, 0.0000, 1.0000]}, 
{"Name": "bias.reverb", "IsOn": false, "Parameters":[0.4810, 0.2030, 0.3152, 0.3607, 0.6429, 0.3393, 0.4000, 0.0000]}],
"Filler": "56"
}
Button pressed: 26
JSON:
{"PresetNumber": 127, "UUID": "A96E8DD2-4C7A-4BC8-8195-1FEECA4D34E6", 
"Name": "Hendrix ", "Version": "0.7", "Description": "Purple Haze", "Icon": "icon.png", "BPM": 120.0000, 
"Pedals": [
{"Name": "bias.noisegate", "IsOn": true, "Parameters":[0.8193, 1.0000]}, 
{"Name": "LA2AComp", "IsOn": false, "Parameters":[0.0000, 0.7126, 0.7248]}, 
[... 404 lines cut ...]
01FE000041FF6A0000000000000000003C187E2803114A3F191B191A04114A0000F7F0010F020301200F0C1900002B62006961732E72657658657262431700115D4A3E2B605F0111F7F0010F500301480F0D194A3F24500C1402114A3F05532C2603114A3E10062D0E04
Sending data:
01FE000041FF33000000000000000000114A3E7062F7F0010F610301600F0E104E05114A623E69374C06114A1C3F19191A2AF7
0xffc1: onWrite()
Received data:
01FE000053FE3C000000000000000000F0011003020100000300000000000000000000000000000000000000000000000000000000000000000000F7
Reading request from Amp
Found request for current preset
Sending data:
01FE000041FF6A000000000000000000F00110630301200F001901005924004139364538444400322D344337412D00344243382D3831F7F00110780301000F011939352D3100464545434134441033344536284865406E647269782023F7F00110000301400F0219302E
Sending data:
01FE000041FF6A000000000000000000372B00507572706C65201048617A65286963406F6E2E706E674AF7F00110350301100F03194270000003172E626961732E006E6F69736567616C7465431200114AF7F00110770301600F04193F513D251601114A3F0000000128
Sending data:
01FE000041FF6A0000000000000000004C413241436F6C6D70421300114AF7F00110320301000F0519000000004601114A3F366C742602114A3F390C7E612446757A7A4312F7F001102F0301300F061900114A3F301E560401114A3F08286C222A4F7261006E67654144
Sending data:
01FE000041FF6A0000000000000000003330F7F001103D0301580F071943150011594A3F19191A0111414A3F3333330211414A3F08240B0311F7F001103A0301080F08194A3F6E5C0D1204114A3F414102202C43686F72750073416E616C6F67F7F00110430301580F09
Sending data:
01FE000041FF6A0000000000000000001943140011594A3E2D2B1F0111594A3E4C4C4D0211414A3F3333330311F7F001104A0301480F0A194A3F4C4C034D2944656C6179304D6F6E6F4215000B114A3E437B4A01F7F001104A0301580F0B19114A3E4B1B6D7A02114A3F
Sending data:
01FE000041FF6A0000000000000000003C187E2803114A3F191B191A04114A0000F7F00110020301200F0C1900002B62006961732E72657658657262431700115D4A3E2B605F0111F7F00110500301480F0D194A3F24500C1402114A3F05532C2603114A3E10062D0E04
Sending data:
01FE000041FF33000000000000000000114A3E7062F7F00110610301600F0E104E05114A623E69374C06114A1C3F19191A2AF7
Address: 6f:21:50:27:fb:4f Unsubscribed to 0xffc2
Client disconnected
Start advertising

stangreg, 2023-03-01:
That is really strange. It processes all initialization steps correctly, only in the last step the app keeps asking for the current preset and does not seem to accept the response from Ignitron.
I assume you have built from the main branch, correct?
I might find some time over the weekend to look into the code, currently I have no idea why it does not work.
Are you running the latest version of the app on your iPhone?
Also, to reproduce can you please send me the exact steps what buttons you press in the app from start to end and also what exactly is shown in the display?

Storgio, 2023-03-01:
What does it mean "built from the main branch"? 
I built everything just following your tutorial..
I'm running last version on my iphone
I click the burger menu, add amp, add Spark 40, connect.
Then the app says "connecting" till it says "disconnected"
The bluetooth icon on ignitron turns on and turns off as soon as I got the "disconnected" message on mobile app.

Should I try to mod the preset list and delete everything?

Thanks again for you help and your work

stangreg, 2023-03-01:
Without having checked the code, you could try to exchange the first preset entry in PresetList.txt with another name and re-upload the data.

Storgio, 2023-03-01:
I will, thanks

Storgio, 2023-03-02:
I tried and I delete all the preset, I uploaded a new Preset.txt and only one preset but nothing really changed.
Once I managed to connect and upload a preset but them the app disconnected and I wasn't anymore able to upload.
It remains connected few moments while is generating this error:

Found request for current preset
Sending data:
01FE000041FF6A000000000000000000F0011B6B0301200F001901005924004542364535314600342D443630462D00343945382D3943F7F0011B250301000F011944312D34003043323936324110384233382541630820446323302E37F7F0011B0E0301080F02193441
Sending data:
01FE000041FF6A00000000000000000043440063202D20426163006B20696E20426C0861636B2869636FF7F0011B1D0301000F03196E2E706E4A674A4270000017012E626961732E6E006F697365676174F7F0011B410301300F0419654312000B114A3E0A0903014B11
Sending data:
01FE000041FF6A0000000000000000004A3F00000028004C413241436F6DF7F0011B6B0301300F05197042130003114A000000000123114A3F366C740253114A3F390C7E27F7F0011B040301000F0619426F6F735874657242110011214A3F445A1D2550306C65786943
Sending data:
01FE000041FF6A0000000000000000001500F7F0011B0C0301180F0719114A3F3518460B01114A3F5D1A424402114A3F6119165303114A3F19F7F0011B7D0301580F0819191A0411294A3F4C607627544072656D6F6C6F422D1300114A3E687FF7F0011B010301680F09
Sending data:
01FE000041FF6A000000000000000000194C01114A683F332C2502114A1C3F182010294465006C61794D6F6E6FF7F0011B7B0301580F0A1942150011454A3E437B4A01115D4A3E4B6D7A0211414A3F3C7E280311F7F0011B610301480F0B194A3F19190D1A04114A0000
Sending data:
01FE000041FF6A0000000000000000000002002B626961732E4072657665726243F7F0011B350301680F0C191700114A6E3E2B605F01114A643F24501402114A603F05532603114AF7F0011B3C0301500F0D193E10060E1604114A3E70624E1605114A3E69374C660611
Sending data:
01FE000041FF220000000000000000004A3F19191AF7F0011B5E0301080F0E0156F7
Address: 7f:38:44:e4:ca:c4 Unsubscribed to 0xffc2

Then it disconnects.
With Android it just works fine.
Thanks again for you help

stangreg, 2023-03-02:
I will try to replicate this at my end in the next days and will get back to you.
Good that it works with Android for you.

Storgio, 2023-03-04:
Ok, I fixed it.
I flashed again the board as a Esp32vrooom 4m with 2mb for app and 2mb for spiffs and now I can upload all your presets (before I had a available space error) and iOS connections works flawless.
I guess it was a partitioning error on my board that was causing some kind of read/write error.
The pedal is just fantastic, I will try to use it with Airstep just controlling looping funcion and it’s going to be just perfect.

thanks for your amazing job.

stangreg, 2023-03-04:
That's great to hear, I am glad thatcher works now!
Thank you for your kind words, I am actually quite proud of the project.
I myself use it on a nearly daily basis, but for me it feels great to hear that others also have fun with it!

I will close this issue now.

closed 2023-03-04 by stangreg

Storgio, 2023-03-05:
You have to be proud, this pedal is great and much better than any commercial one. I added battery support with a 18650.
It could be better with a longer box with all the switches on one line but… I love it anyway.
Once again: THANKS. 🙂

stangreg, 2023-03-05:
Looking at a previous comment, I had a look at the looper functionality of the AirStep.
When I find some time, I could implement similar behavior as I think it sounds pretty useful.

Storgio, 2023-03-05:
The Airstep is a good tool but as a looper it lacks of some features.
It would be great to add the chance to stop and play banks separately without any long pressure.
The best would be to have two separate pedals, one only for controlling the loop functions and the other just for sounds.
I’m using right now Ignitron and Airstep together  and I love the combo.
For looping I would love to have 6 buttons:
Rec/overdub, stop/play session, cancel current track, mute/unmute track, track up, track down.

what do you think?

stangreg, 2023-03-08:
@Storgio I have now updated the code to have a proper Looper functionality. In Looper mode (enter by long pressing Preset 4 button in APP mode), all 6 buttons can now be used for Looper functionality. I have tested it with Loopy HD and the proposed functions you mentioned above and it works perfectly.
You can also assign one of the buttons to the "tap tempo" function, I have also tested this.
You can still switch your presets while in Looper mode by long pressing 'Bank down' and 'Bank up' buttons. Presets will also be switched across banks. If you want to change presets more quickly, you can easily switch back to "normal" mode by long pressing 'Preset 4' button and switching to the desired preset and then go back to Looper mode.

If you want, try it out and let me know what you think!

Storgio, 2023-03-09:
Hi, that’s simply great!In my opinion the best thing would be to build another pedal just for looping, without any other functions: six buttons + six long pressing one, working as a Bluetooth keyboard.No led, all the buttons in a line, no display.I will try to design a pcb to install everything on board using cheap micro switch and actuators, I would run it by battery (I do the same for ignitron).Right now I’m using the Airstep as a looper foot switch connected to my mac m1 running loopy and audio out on monitors and I love it, having six buttons working would be just amazing! Undo function would be very useful too.I will test the new software later but I’m sure it’s already perfect!Inviato da iPhone

stangreg, 2023-03-10:
@Storgio As it was pretty straight forward to implement, I have now added a **Keyboard only** mode. You can enter it by holding `Preset 3` during startup. This will turn the Ignitron into a BLE keyboard without any Spark controlling functionality. It will send characters "A" - "F" with a short press, and "1" - "6" with a long press. You can configure these buttons to any configuration you like in the looper app.
Hope you have fun with this, I think for now I am done with this. If there is anything else, please open a new issue.

Storgio, 2023-03-10:
That's just awesome, many thanks!
I'm developing a 3d printable enclosure inspired to Airstep with optional display support and six buttons and a new pcb to reduce off board wiring.
If you are interested as soon as I'm done i will send it to you.
Thanks again for the amazing job.