stangreg/Ignitron #20: Compilation error: 'stoi' was not declared in this scope

MrBarnickle, 2023-07-20:
Hello,
I have a problem when I try to compile the program. The error I get is:
Compilation error: 'stoi' was not declared in this scope

C:\Arduino\Ignitron\src\SparkDataControl.cpp: In member function 'int SparkDataControl::init(int)':
C:\Arduino\Ignitron\src\SparkDataControl.cpp:67:29: error: 'stoi' was not declared in this scope
   sparkModeInput = stoi(line);
                             ^
C:\Arduino\Ignitron\src\SparkDataControl.cpp:97:30: error: 'stoi' was not declared in this scope
    currentBTMode_ = stoi(line);

Arduino IDE Version 2.1.1
WEMOS LOLIN32

Can someone help me?

Thanks!!

stangreg, 2023-07-20:
@MrBarnickle you have to make sure to use C++11 when compiling.
Please see here: https://stackoverflow.com/questions/22084783/function-stoi-not-declared

MrBarnickle, 2023-07-20:
@stangreg:
from my platform.txt
compiler.cpp.flags=-std=gnu++11 -Os -g3 -Wpointer-arith -fexc....
Output from __cplusplus: 201103

stangreg, 2023-07-20:
@MrBarnickle you can try if the suggested fix to change from stoi call to atoi call as suggested further down that site. 
I can only check myself next week as I don't have access to my environment at the moment.
If you try, let me know if it works.

MrBarnickle, 2023-07-20:
@stangreg:
Thanks!!! atoi solved this problem.

closed 2023-07-20 by MrBarnickle

ServError, 2023-07-20:
Just thought I'd throw in a side note/alternative that's not C++11 dependent. Since sparkModeInput only uses a single character and currently won't exceed a value of '9', you can replace the stoi calls for it with

`(int)(line[0]-'0')`

This leaves a potential hole down the road, but it is functional in the code's current state.