stangreg/Ignitron #23: Missing Ext_nimble_config.h

Bonekrovvn, 2023-09-18:
Hello stangreg,
I'm done with hardware and my friend is doing software side, but he has problem with this missing library **Ext_nimble_config.h**.
Could You please help me what is missing or what shoudl we do?
Thank You! Vit, Czech republic
[image]

stangreg, 2023-09-18:
Hi @Bonekrovvn, 
sorry I cannot read from your description what the exact error is. I am not aware of any required file called "Ext_nimble_config.h". Is there any way you can show me where exactly you face the issue? Do you have any compiler output or other error message you could add to this issue?

Ted666point6, 2023-09-25:
[image]
Hello,
this is that problem what Bonekrovvn wrote. We are not sure where we can find this library.
Thanks for help.

stangreg, 2023-09-25:
Thanks for the additional information. On which platform are you building and which IDE are you using? 
I have checked the code of the NimBLE library and it shows the following in the file "nimconfig.h":

#ifdef ESP_PLATFORM
#include "sdkconfig.h"
#else
#include "ext_nimble_config.h"
#endif

As I am building for an ESP32, the variable ESP_PLATFORM is defined, which means that the line '#include "ext_nimble_config.h"' is NOT executed in my case. Can you please check also if you have configured your IDE to build for the right platform? The README.md file should describe it in detail in the software building part. Maybe something was missed here?

Bonekrovvn, 2023-09-25:
It is this kit:
https://www.laskakit.cz/iot-esp-32s-2-4ghz-dual-mode-wifi-bluetooth-rev-1--cp2102/

stangreg, 2023-09-25:
@Bonekrovvn  It looks good from what I can see. Have you double checked that you configured the board correctly in the IDE? I would assume that this would determine the variable to be set correctly.

stangreg, 2023-10-22:
This issue will be closed soon due to inactivity. I assume it worked in the end?!

Bonekrovvn, 2023-10-23:
Hello @stangreg, I am completely lost with any progress, do You think it would be possible to connect with remote desktop to my laptop and check what is wrong and possibly compile and upload code? I was able to run some primitive stuff like LED diode light but this project is way above my skill...

Thank You.

stangreg, 2023-10-23:
@Bonekrovvn we could do this, I just need to sort out when I can take time for this.
In the meantime, you could send the following information:

- which IDE are you using, Arduino IDE or Sloeber?
- in case of Arduino IDE, which version? 1.8 or 2.x? 
- case Sloeber: go to "Project"->"Properties", then select "Sloeber" on the left. Make screenshot.
- case Arduino: select Tools menu, make screenshot.

Attach the screenshots to this issue, then I can maybe see already if it is configured correctly.

Bonekrovvn, 2023-10-23:
[image]

stangreg, 2023-10-23:
Just a shot in the dark, but have you checked that you chose the right board? You could try other boards with a ESP32-WROOM-32 module, for example "Expressif ESP32 Dev Module". Maybe try a few other fitting ones.

stangreg, 2023-10-24:
@Bonekrovvn another thing you could try if the above does not work is to add the compile flag "-DESP_PLATFORM" in the build configuration.

Bonekrovvn, 2023-10-24:
I tried some examples and uploaded oled display test sketch and it works so I assume that board is set up correctly.
[image]

stangreg, 2023-10-24:
Have you tried setting the compiler flag as I described in my last comment? For the NimBLE library to compile the flag has to be set. It seems that through the Bord config, it is not set, so you could try to set it manually in the IDE to get it compiled.
Your example is probably not relying on that flag to be set.

Bonekrovvn, 2023-10-24:
Sorry, I have no idea how to add compiler flag... :-(

stangreg, 2023-10-24:
Please check here, the first answer has the information you need.

https://forum.arduino.cc/t/additional-compiler-options/631297/2

Summary:
- find the platform.txt for your board (check the folder name)
- create the file platform.local.txt in the same folder
- open the file and add the line 

compiler.cpp.flags=-DESP_PLATFORM -DUSE_NIMBLE

Save the file and try to recompile.
If it does not work, double check that you are in the right folder. For each board configuration, there is a separate platform.txt in the respective folder.
If the folder is correct and it still does not work, you can also try to search the file platform.txt for the pattern "compiler.cpp.flags" and add the two flags as mentioned above.

If it does not work, check if the error message you get is still the same or a different one.

Hope this helps..

Bonekrovvn, 2023-10-25:
[image]

stangreg, 2023-10-25:
I think that is not the right folder to put the platform.local.txt file. Please move it up two levels, into the folder "2.0.11". there should already be a platform.txt, the local file must be in the same folder. Then try again to compile.

Note: The error you sent now is related to the USE_NIMBLE flag, so it seems you don't get the initial error anymore.

Bonekrovvn, 2023-10-25:
In file included from C:\Users\vkocmal.PKD\.arduinoIDE\Ignitron-main\Ignitron\src\SparkDataControl.h:16,
                 from C:\Users\vkocmal.PKD\.arduinoIDE\Ignitron-main\Ignitron\src\SparkButtonHandler.h:13,
                 from C:\Users\vkocmal.PKD\.arduinoIDE\Ignitron-main\Ignitron\Ignitron.ino:11:
C:\Users\vkocmal.PKD\.arduinoIDE\Ignitron-main\Ignitron\src\SparkBLEKeyboard.h:11: warning: "USE_NIMBLE" redefined
 #define USE_NIMBLE
 
<command-line>: note: this is the location of the previous definition
In file included from C:\Users\vkocmal.PKD\.arduinoIDE\Ignitron-main\Ignitron\src\SparkDataControl.h:21,
                 from C:\Users\vkocmal.PKD\.arduinoIDE\Ignitron-main\Ignitron\src\SparkButtonHandler.h:13,
                 from C:\Users\vkocmal.PKD\.arduinoIDE\Ignitron-main\Ignitron\Ignitron.ino:11:
C:\Users\vkocmal.PKD\.arduinoIDE\Ignitron-main\Ignitron\src\SparkStreamReader.h:107:64: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
  void add_str(char* a_title, std::string a_str, char* nature = "all");
                                                                ^~~~~
C:\Users\vkocmal.PKD\.arduinoIDE\Ignitron-main\Ignitron\src\SparkStreamReader.h:108:57: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
  void add_int(char* a_title, int an_int, char* nature = "all");
                                                         ^~~~~
C:\Users\vkocmal.PKD\.arduinoIDE\Ignitron-main\Ignitron\src\SparkStreamReader.h:109:62: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
  void add_float(char* a_title, float a_float, char* nature = "all");
                                                              ^~~~~
C:\Users\vkocmal.PKD\.arduinoIDE\Ignitron-main\Ignitron\src\SparkStreamReader.h:110:52: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
  void add_float_pure(float a_float, char* nature = "all");
                                                    ^~~~~
C:\Users\vkocmal.PKD\.arduinoIDE\Ignitron-main\Ignitron\src\SparkStreamReader.h:111:62: warning: ISO C++ forbids converting a string constant to 'char*' [-Wwrite-strings]
  void add_bool(char* a_title, boolean a_bool, char* nature = "all");
                                                              ^~~~~
c:/users/vkocmal.pkd/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/users/vkocmal.pkd/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/lib\libc.a(lib_a-stdio.o):(.literal+0x20): warning: _close_r is not implemented and will always fail
c:/users/vkocmal.pkd/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/users/vkocmal.pkd/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/lib\libc.a(lib_a-fseeko.o):(.literal+0x3c): warning: _fstat_r is not implemented and will always fail
c:/users/vkocmal.pkd/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/users/vkocmal.pkd/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/lib\libc.a(lib_a-raise.o):(.literal+0x0): warning: _getpid_r is not implemented and will always fail
c:/users/vkocmal.pkd/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/users/vkocmal.pkd/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/lib\libc.a(lib_a-raise.o):(.literal+0x4): warning: _kill_r is not implemented and will always fail
c:/users/vkocmal.pkd/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/users/vkocmal.pkd/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/lib\libc.a(lib_a-stdio.o):(.literal+0xc): warning: _lseek_r is not implemented and will always fail
c:/users/vkocmal.pkd/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/users/vkocmal.pkd/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/lib\libc.a(lib_a-fopen.o):(.literal+0x18): warning: _open_r is not implemented and will always fail
c:/users/vkocmal.pkd/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/users/vkocmal.pkd/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../. [... cut ...]
c:/users/vkocmal.pkd/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0/../../../../xtensa-esp32-elf/bin/ld.exe: c:/users/vkocmal.pkd/appdata/local/arduino15/packages/esp32/tools/xtensa-esp32-elf-gcc/esp-2021r2-patch5-8.4.0/bin/../lib/gcc/xtensa-esp32-elf/8.4.0\libgcc.a(unwind-dw2-fde.o): in function `_Unwind_Find_FDE':
/builds/idf/crosstool-NG/.build/HOST-x86_64-w64-mingw32/xtensa-esp32-elf/src/gcc/libgcc/unwind-dw2-fde.c:1078: undefined reference to `pthread_mutex_unlock'
collect2.exe: error: ld returned 1 exit status

exit status 1

Compilation error: exit status 1

stangreg, 2023-10-25:
There is a lot of undefined references in the compile which should not come if all paths are set up correctly.
Can you please try to do a clean build (i.e. delete the compiled files first) and rebuild again?

Bonekrovvn, 2023-10-25:
I think we are wasting time because I have no idea what to do...

stangreg, 2023-10-25:
I could actually replicate the same compiling errors in my Arduino IDE. Usually I am using Sloeber as the IDE which works fine. I will have another look what is causing the failure in the Arduino IDE. 
In the meantime you could try to switch to Slobber (which gives some other benefits), but I will try to get the build running on Arduino IDE (which strangely worked the last time I tried a few months ago).

Bonekrovvn, 2023-10-25:
Probably all those problems are caused by my noob approach to IDE so there is something missing for sure. I downloaded and run Sloeber but it is even more complicated than Arduino IDE...

stangreg, 2023-10-25:
As mentioned, I could reproduce the issue in my Arduino IDE. It happens even with the minimal example file. 
I suspect either the Arduino version, or the board version acting strange (although Sloeber uses the same board version, afaik).

Sloeber looks more complicated in the beginning, but basically you just need to create a new project with the existing files, configure the board and select the libraries (it uses the same as Arduino IDE).
Just note that some settings can be found in the Sloeber menu item and some in the project properties menu after creating the project.
Maybe you can give it a try, Google might also help to set it up.
In case I find the reason for the Arduino build to fail, I will comment here.

Bonekrovvn, 2023-10-26:
[image]
[image]
[image]
[image]

stangreg, 2023-10-26:
OK, I just tried to set up Ignitron as a new project in Slobber, these are the steps I took:

- File -> New -> Arduino Sketch (this should create a new empty sketch)
- Name the Project "Ignitron"
- Copy the cloned local git folder into the workspace (don't use the Import Wizard)
- Refresh the project in Sloeber. The structure should look similar to this:
[image]
- Select the options and libraries similar to what you have already done in the screenshots above. You might need to select the library "SPIFFS". Library "FreeRTOS" does not need to be selected.
- Try to build the workspace.

If it does not work, you can also try to create a new Empty test project and try to compile after setting the board config.

stangreg, 2023-10-26:
If you still get an error, please post the full log output here as well

Bonekrovvn, 2023-10-26:
[image]
13:19:04 **** Build of configuration Release for project Ignitron ****
"C:\\Sloeber\\arduinoPlugin\\tools\\make\\make" all 
C:/Sloeber/arduinoPlugin/tools/make/make --no-print-directory pre-build
cmd /c if exist "C:\sloeber-workspace\Ignitron\partitions.csv" COPY /y "C:\sloeber-workspace\Ignitron\partitions.csv" "C:\sloeber-workspace\Ignitron\Release\partitions.csv"
cmd /c if not exist "C:\sloeber-workspace\Ignitron\Release\partitions.csv" if exist "C:\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\2.0.11\variants\esp32dapartitions.csv" COPY "C:\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\2.0.11\variants\esp32dapartitions.csv" "C:\sloeber-workspace\Ignitron\Release\partitions.csv"
cmd /c if not exist "C:\sloeber-workspace\Ignitron\Release\partitions.csv" COPY "C:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/2.0.11\tools\partitionsdefault_8MB.csv" "C:\sloeber-workspace\Ignitron\Release\partitions.csv"
Syst‚m nem…§e nal‚zt uvedeně soubor.
makefile:138: recipe for target 'pre-build' failed
make[1]: *** [pre-build] Error 1
makefile:98: recipe for target 'all' failed
make: *** [all] Error 2
"C:/Sloeber/arduinoPlugin/tools/make/make all" terminated with exit code 2. Build might be incomplete.

13:19:05 Build Failed. 4 errors, 0 warnings. (took 1s.286ms)

stangreg, 2023-10-26:
The line "Syst‚m nem…§e nal‚zt uvedeně soubor." looks like partially czech and partially broken (maybe due to special characters in the message). Do you see the line different in the application? 
Maybe you could try to set the Slobber language to English (if not already done)?

stangreg, 2023-10-26:
Another thing you could try is to change the board version in Sloeber to 2.0.9. You can do this in the menu "Sloeber" -> "Preferences", then select Platform and Boards and deselect the current selected board and select the version 2.0.9. Depending on your setup, you need to add the following URLs first in the item "Third party index URLs": 

https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json
https://dl.espressif.com/dl/package_esp32_index.json

Then the versions should be present in the Board menu.

Bonekrovvn, 2023-10-26:
> The line "Syst‚m nem…§e nal‚zt uvedeně soubor." looks like partially czech and partially broken (maybe due to special characters in the message). Do you see the line different in the application? Maybe you could try to set the Slobber language to English (if not already done)?

It says something like: system cannot find listed file

Sloeber looks like it has default english and I wasn´t able to find where to change it.

Bonekrovvn, 2023-10-26:
> Another thing you could try is to change the board version in Sloeber to 2.0.9. You can do this in the menu "Sloeber" -> "Preferences", then select Platform and Boards and deselect the current selected board and select the version 2.0.9. Depending on your setup, you need to add the following URLs first in the item "Third party index URLs":
> 
> https://raw.githubusercontent.com/espressif/arduino-esp32/gh-pages/package_esp32_index.json https://dl.espressif.com/dl/package_esp32_index.json
> 
> Then the versions should be present in the Board menu.
> 
> To be on the safe side, select "Project" -> "Clean..." and confirm to do a clean build.
[image]

stangreg, 2023-10-26:
Ok the version change, you have to select/install it first in the menu Arduino->Preferences.
It seems that the menu is called Arduino on Windows, on Mac it is called Sloeber.

Bonekrovvn, 2023-10-26:
I had to restart Sloeber then 2.0.9 appeared but still that same error message:

19:09:08 **** Build of configuration Release for project Ignitron ****
"C:\\Sloeber\\arduinoPlugin\\tools\\make\\make" all 
C:/Sloeber/arduinoPlugin/tools/make/make --no-print-directory pre-build
cmd /c if exist "C:\sloeber-workspace\Ignitron\partitions.csv" COPY /y "C:\sloeber-workspace\Ignitron\partitions.csv" "C:\sloeber-workspace\Ignitron\Release\partitions.csv"
cmd /c if not exist "C:\sloeber-workspace\Ignitron\Release\partitions.csv" if exist "C:\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\2.0.9\variants\esp32dapartitions.csv" COPY "C:\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\2.0.9\variants\esp32dapartitions.csv" "C:\sloeber-workspace\Ignitron\Release\partitions.csv"
cmd /c if not exist "C:\sloeber-workspace\Ignitron\Release\partitions.csv" COPY "C:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/2.0.9\tools\partitionsnoota_ffat.csv" "C:\sloeber-workspace\Ignitron\Release\partitions.csv"
Syst‚m nem…§e nal‚zt uvedeně soubor.
makefile:138: recipe for target 'pre-build' failed
make[1]: *** [pre-build] Error 1
makefile:98: recipe for target 'all' failed
make: *** [all] Error 2
"C:/Sloeber/arduinoPlugin/tools/make/make all" terminated with exit code 2. Build might be incomplete.

19:09:10 Build Failed. 4 errors, 0 warnings. (took 1s.296ms)

stangreg, 2023-10-26:
Can you see the file partitions.csv in the Release folder? If not, maybe you can try a different partition schema, Default for example.

Bonekrovvn, 2023-10-26:
There is no such file in release folder, after trying several different partition schemes still same error.

stangreg, 2023-10-27:
OK, I think I found something. There seems to be an issue with Sloeber with ESP versions greater than 2.0.10.
I had a similar issue as you (file not found) when trying to use version 2.0.14. 

The issue is described [here](https://github.com/Sloeber/arduino-eclipse-plugin/discussions/1576) and I could successfully build after applying the suggested solution. Let me summarize for you:

- Open the folder "C:/Sloeber/arduinoPlugin/packages/esp32/hardware/esp32/2.0.14"
- Delete the file "platform.sloeber.txt"
- Open the file "platform.txt" with a text editor and add the following lines

> recipe.hooks.prebuild.7.pattern=bash -c "[ -f "{build.path}"/file_opts ] || touch "{build.path}"/file_opts"
> recipe.hooks.prebuild.7.pattern.windows=cmd /c if not exist "{build.path}\file_opts" type nul > "{build.path}\file_opts"

You can add them anywhere (I assume), I just put them near the other lines starting with "recipe.hooks".

- Save and close the file platform.txt
- Open the Ignitron project properties in Sloeber
- Check that the folder mentioned above is selected for the board configuration (2.0.14) and click "Apply and Close". Even if you are sure you have the right board, you need to open the properties and click "Apply and Close". This should trigger the recreation of the platform.sloeber.txt file.
- Check if the file platform.sloeber.txt is created and contains one of the two lines. For Windows systems, only the second line should be in the file, for Linux systems the first.
- Delete the build folder "Release" in the project Workspace. This will force Sloeber to rebuild the makefiles.
- Build

Let me know if it works and what the log output is. Please send the whole output in case of failure.

Bonekrovvn, 2023-10-27:
> OK, I think I found something. There seems to be an issue with Sloeber with ESP versions greater than 2.0.10. I had a similar issue as you (file not found) when trying to use version 2.0.14.
>
> The [...]

Tried several times, still same... 

18:17:45 **** Incremental Build of configuration Release for project Ignitron ****
"C:\\Sloeber\\arduinoPlugin\\tools\\make\\make" all 
C:/Sloeber/arduinoPlugin/tools/make/make --no-print-directory pre-build
cmd /c if exist "C:\sloeber-workspace\Ignitron\partitions.csv" COPY /y "C:\sloeber-workspace\Ignitron\partitions.csv" "C:\sloeber-workspace\Ignitron\Release\partitions.csv"
cmd /c if not exist "C:\sloeber-workspace\Ignitron\Release\partitions.csv" if exist "C:\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14\variants\esp32dapartitions.csv" COPY "C:\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14\variants\esp32dapartitions.csv" "C:\sloeber-workspace\Ignitron\Release\partitions.csv"
cmd /c if not exist "C:\sloeber-workspace\Ignitron\Release\partitions.csv" COPY "C:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/2.0.14\tools\partitionsdefault.csv" "C:\sloeber-workspace\Ignitron\Release\partitions.csv"
Syst‚m nem…§e nal‚zt uvedeně soubor.
makefile:138: recipe for target 'pre-build' failed
make[1]: *** [pre-build] Error 1
makefile:98: recipe for target 'all' failed
make: *** [all] Error 2
"C:/Sloeber/arduinoPlugin/tools/make/make all" terminated with exit code 2. Build might be incomplete.

18:17:46 Build Failed. 4 errors, 0 warnings. (took 1s.296ms)

stangreg, 2023-10-27:
The log starts with incremental build. Can you please remove the Release folder and do a clean build? Then send the log file.

Bonekrovvn, 2023-10-27:
18:35:58 **** Build of configuration Release for project Ignitron ****
"C:\\Sloeber\\arduinoPlugin\\tools\\make\\make" all 
C:/Sloeber/arduinoPlugin/tools/make/make --no-print-directory pre-build
cmd /c if exist "C:\sloeber-workspace\Ignitron\partitions.csv" COPY /y "C:\sloeber-workspace\Ignitron\partitions.csv" "C:\sloeber-workspace\Ignitron\Release\partitions.csv"
cmd /c if not exist "C:\sloeber-workspace\Ignitron\Release\partitions.csv" if exist "C:\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14\variants\esp32dapartitions.csv" COPY "C:\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14\variants\esp32dapartitions.csv" "C:\sloeber-workspace\Ignitron\Release\partitions.csv"
cmd /c if not exist "C:\sloeber-workspace\Ignitron\Release\partitions.csv" COPY "C:\Sloeber\/arduinoPlugin/packages/esp32/hardware/esp32/2.0.14\tools\partitionsdefault.csv" "C:\sloeber-workspace\Ignitron\Release\partitions.csv"
Syst‚m nem…§e nal‚zt uvedeně soubor.
makefile:138: recipe for target 'pre-build' failed
make[1]: *** [pre-build] Error 1
makefile:98: recipe for target 'all' failed
make: *** [all] Error 2
"C:/Sloeber/arduinoPlugin/tools/make/make all" terminated with exit code 2. Build might be incomplete.

18:35:59 Build Failed. 4 errors, 0 warnings. (took 1s.501ms)

Bonekrovvn, 2023-10-27:
> The log starts with incremental build. Can you please remove the Release folder and do a clean build? Then send the log file.
> 
> And have you checked that the additional lines are found innthe platform.sloeber.txt file? 

Both of those lines are present in that txt.

stangreg, 2023-10-27:
I am still confused about the line 
"Syst‚m nem…§e nal‚zt uvedeně soubor." That looks really strange, not like a proper message. Does that look correct to you (because of language) or does it also look strange to you?

Also, I would have expected to have only one line in the platform.sloeber.txt file.

Can you confirm that you have executed the steps exactly as described above? I.e. edit the platform.txt, not the platform.sloeber.txt.

Previously I had also an error about file not found, but it was more specific on the file it did not found. 
I can see no specific error message in your build, really strange..

Bonekrovvn, 2023-10-27:
> Does the message mean something like access denied or like file not found?

It goes: system cannot find the specified file

Bonekrovvn, 2023-10-27:
> I am still confused about the line "Syst‚m nem…§e nal‚zt uvedeně soubor." That looks really strange, not like a proper message. Does that look correct to you (because of language) or does it also look strange to you? 
Little bit of font failure because of specific czech letters.
> 
> Also, I would have expected to have only one line in the platform.sloeber.txt file.
> 
> Can you confirm that you have executed the steps exactly as described above? I.e. edit the platform.txt, not the platform.sloeber.txt.
 I did several times.
> 
> Previously I had also an error about file not found, but it was more specific on the file it did not found. I can see no specific error message in your build, really strange..

stangreg, 2023-10-27:
> > Does the message mean something like access denied or like file not found?
> 
> 
> 
> It goes: system cannot find the specified file

It would be interesting to know which file it is missing.
Is there any way to switch on a more verbose logging?

Bonekrovvn, 2023-10-27:
I am not sure how to get more complex log - I copied those above from build console.

Bonekrovvn, 2023-10-27:
[image]

stangreg, 2023-10-27:
The screenshot is unrelated to the build.

I am running out of ideas here. For me it works fine.
You could try to compile it for "Node32s" board, the it would be the same as for me.

Another alternative would bea complete reinstall (including Arduino IDE uninstall), there is something strange here.

Bonekrovvn, 2023-10-27:
I tried for few different versions of boards but that error remains same.
Should I try to install older version of Arduino IDE and Sloeber?

stangreg, 2023-10-27:
> I tried for few different versions of boards but that error remains same.
> 
> Should I try to install older version of Arduino IDE and Sloeber?

Unwound actually completely uninstall both Sloeber and Arduino IDE, delete the folder C:/Sloeber,   Delete the workspace folder, delete the cloned GitHub directory, then maybe even restart the machine.

Then install Arduino IDE again, Sloeber again, clone the repository and try to setup from scratch.

Bonekrovvn, 2023-10-27:
After reinstalling everything still that same error message so I must do some basic mistake somewhere...

stangreg, 2023-10-27:
Not giving up yet...

I compared the steps from your build process with mine to see where it might fail for you.

In the beginning, the build process tries to copy some files into the Release folder depending on where it finds it first. It will first look into the project folder, then the esp32 config folders.

Can you please check if the following files exist in the given locations on your system:

- C:\sloeber-workspace\Ignitron\partitions.csv
- C:\sloeber-workspace\Ignitron\Release\partitions.csv
- C:\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14\variants\esp32da\partitions.csv
- C:\Sloeber/arduinoPlugin/packages/esp32/hardware/esp32/2.0.14\tools\partitions\default.csv
- C:\sloeber-workspace\Ignitron\bootloader.bin
- C:\sloeber-workspace\Ignitron\src\bootloader.bin
- C:\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14\variants\bootloader.bin
- C:\Sloeber\arduinoPlugin\packages\esp32\tools\esptool_py\4.5.1\esptool

Let me know which ones you find and which ones not.

What is strange is that in some paths in your build log are missing a "\\", for example here: 
C:\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14\variants\esp32dapartitions.csv
This should be
C:\Sloeber\arduinoPlugin\packages\esp32\hardware\esp32\2.0.14\variants\esp32da\partitions.csv

So it might be that some variables are not configured correctly in your Sloeber.
Please check in project properties -> C/C++ Build -> Environment for the following:

- The values of the following variables should end with a "\\" . Add it if missing.
  - archive_file_path
  - build.core.path
  - build.opt.path
  - build.path
  - build.source.path
  - build.system.path
  - build.variant.path
  - compiler.path
  - compiler.sdk.path

You can also check if the paths have consistently "\" as a path separator and adjust accordingly to be safe.
When done, please do a clean build (via Project -> Clean...) and check if you get further in the process. 
If so, we can do some more checks.

In addition, please let me know the value of the following variables:
- name
- program.tool
- program.tool.default
- tools.esptool_py.cmd
- upload.tool
- upload.tool.default
- file_opts.path

I have the hope that this brings us at least a bit closer.

Bonekrovvn, 2023-10-28:
I found one strange thing - that same error message appears right after creating new arduino sketch no matter how I name the new project.

stangreg, 2023-10-28:
Yes, that could be because of the incorrect variable settings. I suggest doing what I mentioned above and try again

stangreg, 2023-10-28:
I forgot maybe the most important path variable. Please make sure that the variables

runtime.tools.esptools_py.path

runtime.tools.esptools_py-4.5.1.path

runtime.tools.esptools_py4.5.1.path

have a \ at the end.

stangreg, 2023-11-01:
@Bonekrovvn did you have the chance yet to try the steps I suggested in my last three messages?

Bonekrovvn, 2023-11-03:
Hello, I tried but still same... I recorded new sketch process and I think that problem is at the very start 
https://www.youtube.com/watch?v=c6rqRISNFzY

stangreg, 2023-11-03:
Thanks for the video. I cannot judge if the board config is correct, but that should not matter as the compile is independent of that.

In the video you did not adjust the environment variables in the project properties as described in my last 3 comments. Can you maybe try this and check if the compile at least goes further?

There is one longer comment and then a shorter one because I forgot one important variable.

Bonekrovvn, 2023-11-05:
Should I set those variables before starting new sketch? Because all lines are empty as visible in screenshot.
[image]

After starting new sketch and setting up board options it automaticaly runs those lines as captured in video. Should I ignore it and edit variables after?

Also, I am not sure if I it is ok just to drag and drop Ignitron-main file to project explorer as I always do.

Bonekrovvn, 2023-11-05:
Should it look like this?
[image]
[image]

stangreg, 2023-11-05:
> Should I set those variables before starting new sketch? Because all lines are empty as visible in screenshot.
>
> [image]
>
>
>
> After starting new sketch and setting up board options it automaticaly runs [...]

You need to set the variables after setting up the sketch, before you compile. If it automatically compiles after first setup, no problem, just adjust the variables and rerun the build.

I would recommend trying to build an empty sketch before trying Ignitron, so you can ensure that the build in general works.

However, when you setup the Ignitron I would recommend to start a new sketch and then select the "Import..." command in Project Menu and point to the Ignitron.ino file on your checked out folder.

[2 more replies]

closed 2024-02-13 by stangreg