stangreg/Ignitron #24: BLE connection issues in AMP mode

oskarsjogren, 2023-10-21:
Very cool project, all is working except to connect to my iPhone through the Spark app. It gives me the "Failed to connect" message.

Steps to reproduce the behavior:
1. Start the Ignitron in AMP mode (long pressing switch 1 during boot up)
2. Choose "Connect" in Spark App on iPhone (with Spark Amp off)

The serial monitor in Arduino IDE says this:
======= Entering AMP mode =======
Initialization done.
Multi-connect support: start advertising
Address: 7f:a3:ea:ef:61:d6 Subscribed to notfications for 0xffc2
Address: 7f:a3:ea:ef:61:d6 Unsubscribed to 0xffc2
Client disconnected
Start advertising

The duration for this is 5 s, I have tried to increase to 30 as it says in the comments in "SparkBLEControl.cpp", but to no avail.

Is this linked to the Spark App on the iPhone not supporting the Ignitron anymore, or am I mlissing something?

Thanks
Oskar

stangreg, 2023-10-21:
@oskarsjogren great to hear from another Ignitron build, thanks for your kind words!
Regarding the issue: The app should still work with Ignitron. As it sends multiple requests to Ignitron it would be interesting to see what exactly happens. To do so, please set the DEBUG flag in the file Config.h and rebuilt. The logs should then show the exact messages. Afterwards, I can have a look.

oskarsjogren, 2023-10-22:
Thanks @stangreg .

This morning, flashing with debug flag on, it works...

Output from Serial monitor:
Operation mode: 2
======= Entering AMP mode =======
Initialization done.
Multi-connect support: start advertising
Address: 51:c2:d5:d1:ad:74 Subscribed to notfications for 0xffc2
0xffc1: onWrite()
Received data:
01FE000053FE17000000000000000000F00101000223F7
Reading request from Amp
Found request for serial number
Sending data:
01FE000041FF29000000000000000000F001013F0323020D2D533939394300393939423939390177F7
Clients notified.
0xffc1: onWrite()
Received data:
01FE000053FE17000000000000000000F00102000272F7
Reading request from Amp
Found invalid request: 72 
Clients notified.
0xffc1: onWrite()
Received data:
01FE000053FE17000000000000000000F00103000271F7
Reading request from Amp
Found invalid request: 71 
Clients notified.
0xffc1: onWrite()
Received data:
01FE000053FE17000000000000000000F0010400022FF7
Reading request from Amp
Found request for firmware version
Sending data:
01FE000041FF1D000000000000000000F001046A032F114E01070536F7
Clients notified.
0xffc1: onWrite()
Received data:
01FE000053FE1D000000000000000000F0010515022A011400010203F7
Reading request from Amp
Found request for hw checksum
Sending data:
01FE000041FF1E000000000000000000F001052D032A0D147D4C075A58F7
Clients notified.
0xffc1: onWrite()
Received data:
01FE000053FE17000000000000000000F00106000210F7
Reading request from Amp
Found request for hw preset number
Sending data:
01FE000041FF1B000000000000000000F0010600031000000000F7
Clients notified.
0xffc1: onWrite()
Received data:
01FE000053FE3C000000000000000000F0010701020100010000000000000000000000000000000000000000000000000000000000000000000000F7
Reading request from Amp
Found request for current preset
Sending data:
01FE000041FF6A000000000000000000F001077403012010001901005924004336434338354400362D413646352D00343330322D4146F7F001073103010010011946362D30003036304439393610413342313353770065657420436869F7F00107000301001002196C64
Sending data:
01FE000041FF6A000000000000000000204F4066204D696E652308302E3725536C610473682869636F6EF7F00107660301001003192E706E67614A43340000172E00626961732E6E6F0069736567617465F7F0010719030158100419431200114D4A3E1D597F01113D4A
Sending data:
01FE000041FF6A0000000000000000003E6E3E60284C00413241436F6D70F7F001076803015810051942130011414A000000000111514A3F366C740211294A3F390C7E2D44F7F001073D0301001006196973746F007274696F6E54533639431300114A3E334F037B0111
Sending data:
01FE000041FF6A0000000000000000004A3EF7F001071D030108100719263F14026B114A3E74602A2E004F7665724472690076656E4A4D3435F7F001077C03015810081943150011514A3F3B6A680111594A3F5A5A3A0211494A3F242F4F0311F7F00107250301681009
Sending data:
01FE000041FF6A000000000000000000194A3E1D254C1204114A3F516B0305275472656D6F6C6C6F421300114AF7F0010737030150100A193E687F4C4601114A3F332C256602114A3F182010012944656C61794DF7F001071E030140100B196F6E6F432D1500114A3E07
Sending data:
01FE000041FF6A0000000000000000006C2D0B01114A3E0C636D7102114A3E1607F7F0010708030160100C192B03114A6C3F19191A04114A10000000002B62690061732E72657665F7F0010773030160100D19726243171600114A3E1247455601114A3E51194E160211
Sending data:
01FE000041FF4C0000000000000000004A3E143958F7F0010779030130100E1903114A3E33464F4204114A3F3415040105114A3F3026666606114A3EF7F0010743030130100F044C4C4D25F7
Clients notified.
0xffc1: onWrite()
Received data:
01FE000053FE17000000000000000000F00108000271F7
Reading request from Amp
Found invalid request: 71 
Clients notified.

I really don't know why, but now it works. Forget this issue.

closed 2023-10-22 by oskarsjogren