stangreg/Ignitron #26: BLE Keyboard mode as controller for video tutorial and lessons

itarozzi, 2024-01-24:
I found the BLE keyboard mode useful in those apps where I can remap the connection (Android or Linux in my case) but I would like to use it to also control my video tutorial or lessons. Just to pause/play and go forward/backward without using PC keyboard or touch control every time.

The problem is that different applications use different shortcuts.

For YouTube I found this [link](https://support.google.com/youtube/answer/7631406?hl=en) .
Other web players use other keystroke.
VLC more (in this case maybe I can remap.

In general, the space and left/right arrow keys do the job, in other cases the ctrl or shift arrows to increase the seek time.

I saw the `SparkTypes.h` file to remap the key associated with the buttons. So I have 2 questions:

- How can I send arrow keys? And they could be shift/ctrl combinations 
- And how to display them on display?

If this isn't easy to achieve, I think an alternative solution is to use my Chocolate MIDI pedalboard and (in Linux) use some program to remap MIDI to Keystroke (e.g. [midi2input](https://gitlab.com/enetheru/midi2input )) but I would prefer to use Ignitron

stangreg, 2024-01-24:
@itarozzi Please check the documentation of the [BLE Keyboard library](https://github.com/T-vK/ESP32-BLE-Keyboard/blob/master/README.md). There you can see the codes of the different keys (in the file `BleKeyboard.h`).

Regarding Shift/CTRL combinations, there is one example in the libraries readme. However, I think as you need to use a combination of press/release commands, this might not work natively with the current code of Ignitron. Maybe this information helps you to move forward.

The file `SparkDisplayControl.cpp` handles showing the keyboard presses. However, as I have not tested it with special keys, it might not work with the arrow keys. I assume you would need to defined bitmap data in case you want to show an actual arrow.

itarozzi, 2024-01-24:
@stangreg thanks a lot for usefull informations.

I think I need to refactor a bit the code since you use `keyboardShortPress` and `keyboardShortPress` as str vectors and `print()` method of bleKeyboard. 

Special keys require to use int values and `write()` method.

Not a big problem. I'll think about how to handle this in a simple way and with display modification too.

Thanks

itarozzi, 2024-01-24:
@stangreg I can confirm that it works!  Let me complete my refactory and add a feature I have in mind to manage multiple  keyboard laypout and I will push my branch, FYI

itarozzi, 2024-01-24:
[here](https://github.com/itarozzi/Ignitron/tree/20240124_blekeyboard_improvement
) my branch with a first implementation that seems to work. I need to test better.

stangreg, 2024-01-24:
I have just done a code review from your branch (compared to my main) and it looks pretty straightforward. Good idea with the struct changes.
After you are finished with your test, do you want to create a pull request so I can review the changes and pull into my repository? I also need to adjust the documentation on the build procedure to explain how to configure the keys which should be sent before building.

stangreg, 2024-02-03:
@itarozzi I have now merged and tested the pull request into my main branch. I had to do some adjustments to properly display the last pressed key in the display which should now work.
One idea for a future extension would be to define multiple mappings in an array and use one key in keyboard mode to switch between these mappings (e.g., by long pressing the Bank Up button). This could then cycle through the different mappings required for different apps.

stangreg, 2024-02-13:
@itarozzi I have now added a keyboard manager where you can define multiple keyboards. In KEYBOARD only mode, long pressing `bank down/up` goes to the previous/next keyboard map. It also shows the current mapping on the display.
The `init()` method of the class `SparkKeyboardControl` is used to define the keyboards. When in APP LOOPER mode, the standard keyboard is used, which is the first added to the list. 

A later version could use a (json) config file, for now this might be sufficient.

itarozzi, 2024-02-13:
@stangreg  Congrats!  an  amazing job!

I'm waiting for a new ESP32 board  because I broke mine testing the power-by-battery :)

As soon as it arrives I will try this new version. Thanks again

stangreg, 2024-03-12:
@itarozzi any updates here? I would like to close the issue soon.

itarozzi, 2024-03-12:
Sorry for long delay.
I tested and it work as expected.

Just a minor typo in SparkKeyboardControl.cpp, where you repeated "mapping1.mappingName" for both config. But I can see the mappingName is not yet used.

I think you can close this issue.
Thanks

stangreg, 2024-03-12:
@itarozzi Thanks for the swift reply! 
Great to hear that it works as expected. Funny you mentioned that typo, I coincidentally just ran a code analysis tool and it reported the same. I will fix it soon, and you are right, it is not yet used anywhere.

closed 2024-03-12 by stangreg