stangreg/Ignitron #28: APP mode hanging - incomplete data chunks?

Rorzaborg, 2024-02-09:
Hello, I have a bit of an issue where once booted into APP mode and connected to the Spark, the display shows HW 1 and the preset switches become unresponsive. The serial is repeatedly printing the below, any idea what could be causing this?

Getting current preset from Spark
Sending block:
01FE000053FE3C000000000000000000F0010101020100010000000000000000000000000000000000000000000000000000000000000000000000F7
Done with this command!
Received data:
01FE000041FF6A000000000000000000F00101000438F7F001016A0301200F001901005924003734323532313100372D433241412D00343133352D3846F7F00101190301000F011939322D370043464441303146103531363727312D20436C65616E2330F7F001014203
Received data:
01FE000041FF6A00000000000000000001200F02192E372731402D436C65616E280069636F6E2E706E4A674A4270000017F7F00101090301080F03192E62696100732E6E6F69736530676174654313003B114A3D756E4301F7F00101120301580F0419114A3E29592F12
Last chunk incomplete, ignoring
Received data:
01FE000041FF6A00000000000000000002114A3F000400002A436F6D7040726573736F7243F7F00101260301680F05191200114A6E3E2A3B1001114A143F7F474B27426F606F737465724311F7F001011D0301300F061900114A3F080F1F7824547769366E431500114A
Last chunk incomplete, ignoring
Received data:
01FE000041FF6A0000000000000000003F341D097901114A3EF7F00101460301280F071961740A021B114A3E417054032B114A3E7B13490453114A3F2079532CF7F00101370301000F081943686F72007573416E616C6F3667421400114A3E3541153201114A3FF7F001
Last chunk incomplete, ignoring
Received data:
01FE000041FF6A000000000000000000012C0301000F0919115B1E0223114A3E5D4944034B114A3E000000290044656C61794D6FF7F00101760301600F0A196E6F43156600114A3E1F31206601114A3E6E24611602114A3E7B2457F7F00101590301300F0B1903114A3F
Last chunk incomplete, ignoring
Received data:
01FE000041FF6A000000000000000000341B556A04114A3F090000002B62696100732E7265766572F7F00101080301300F0C19624318000B114A3E2D3027013B114A3E28193B023B114A3E60173203F7F00101240301180F0D19114A3F315B162004114A3E795B797A05
Last chunk incomplete, ignoring
Received data:
01FE000041FF33000000000000000000114A3E6E594A3806114A3E19F7F00101680301580F0E0A191A0711054A3F0000005EF7
Last chunk incomplete, ignoring
Invalid block start, ignoring all data

stangreg, 2024-02-09:
@Rorzaborg thanks for reporting the issue. It might be related to a recent change in the code to support the Spark GO. I will need to look into it in the next days when Infond some time.
In the meantime you could try to check out an earlier version of the main branch, e.g., commit 9a1849e444f5a26598279f65e57f3f6877c9b53f.

stangreg, 2024-02-11:
@Rorzaborg  I have now applied a fix which should make operations with the Spark 40 more stable. Spark GO is still a different challenge as it does not send messages as reliable and in the standard format as the 40 does. Please try to check out the current main branch and rebuild.

Rorzaborg, 2024-02-12:
@stangreg Champion! The issue is completely resolved.

closed 2024-02-12 by Rorzaborg