stangreg/Ignitron #29: Connection to SparkGo issue

thrivewebdesign, 2024-02-18:
When trying to connect to spark go  a reboot loop occurs with the following error log.

21:10:42]Found Spark, connecting.
[21:10:42]Scan ended.
[21:10:42]Connected to: f7:eb:ed:0d:b5:3e
[21:10:42]Subscribing to service notifications of FFC2
[21:10:42]Notifications turned on
[21:10:42]Done with this device.
[21:10:42]BLE connection to Spark established.
[21:10:42]Changing to HW preset 1
[21:10:43]Changing to HW preset 1
[21:10:44]Changing to HW preset 1
[21:10:45]Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
[21:10:45]
[21:10:45]Core  0 register dump:
[21:10:45]PC      : 0x400d310f  PS      : 0x00060e30  A0      : 0x800e5e2c  A1      : 0x3ffd1690  
[21:10:45]A2      : 0x3ffd16c4  A3      : 0xfffffff4  A4      : 0x00000014  A5      : 0x00060023  
[21:10:45]A6      : 0x007bc970  A7      : 0x003fffff  A8      : 0x00000014  A9      : 0x3ffd435f  
[21:10:45]A10     : 0x00000070  A11     : 0x3ffd4328  A12     : 0x3ffd41f8  A13     : 0x3ffd3f68  
[21:10:45]A14     : 0x00000000  A15     : 0x00000003  SAR     : 0x00000020  EXCCAUSE: 0x0000001c  
[21:10:45]EXCVADDR: 0xfffffff4  LBEG    : 0x40090839  LEND    : 0x4009085b  LCOUNT  : 0x00000000  
[21:10:45] 
[21:10:45]
[21:10:45]Backtrace: 0x400d310c:0x3ffd1690 0x400e5e29:0x3ffd16b0 0x400d5f3b:0x3ffd16f0 0x400d6332:0x3ffd1870 0x401e1452:0x3ffd18b0 0x400f7065:0x3ffd18d0 0x400ea0a9:0x3ffd1950 0x400ea0f3:0x3ffd1970 0x400eb50c:0x3ffd1990 0x400e9de5:0x3ffd19f0 0x400e8319:0x3ffd1a10 0x400ef4d3:0x3ffd1a40 0x400edf88:0x3ffd1a70 0x400edf93:0x3ffd1a90 0x400814b6:0x3ffd1ab0 0x400f7bc2:0x3ffd1ad0
[21:10:45]
[21:10:45]
[21:10:45]
[21:10:45]
[21:10:45]ELF file SHA256: 2bc66c482fdf826f
[21:10:45]
[21:10:45]E (5391) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
[21:10:45]Rebooting...
[21:10:45]ets Jun  8 2016 00:22:57
[21:10:45]
[21:10:45]rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
[21:10:45]configsip: 0, SPIWP:0xee
[21:10:45]clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
[21:10:45]mode:DIO, clock div:1
[21:10:45]load:0x3fff0000,len:392
[21:10:45]load:0x40078000,len:14204
[21:10:45]ho 0 tail 12 room 4
[21:10:45]load:0x40080400,len:3236
[21:10:45]entry 0x4008057c
[21:10:45]E (50) esp_core_dump_flasrcore dump partition found!
[21:10:45]E (50) esp_core_dump_flash: No core dump partition found!
[21:10:45]Initializing
[21:10:45]Operation mode (boot): 1
[21:10:45]E (12) SPIFFS: mount failed, -10025
[21:10:45]E (13) SPIFFS: mount failed, -10025
[21:10:45]ERROR while trying to open presets list file
[21:10:45]E (24) SPIFFS: mount failed, -10025
[21:10:45]E (25) SPIFFS: mount failed, -10025
[21:10:45]Stopping advertising keyboard
[21:10:45]Starting scan
[21:10:45]Operation mode: 1
[21:10:45]======= Entering APP mode =======
[21:10:45]Initialization done.
[21:10:47]Found Spark, connecting.
[21:10:47]Scan ended.
[21:10:48]Connected to: f7:eb:ed:0d:b5:3e
[21:10:48]Subscribing to service notifications of FFC2
[21:10:48]Notifications turned on
[21:10:48]Done with this device.
[21:10:48]BLE connection to Spark established.
[21:10:48]Changing to HW preset 1
[21:10:48]Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
[21:10:48]
[21:10:48]Core  0 register dump:
[21:10:48]PC      : 0x400d310f  PS      : 0x00060e30  A0      : 0x800e5e2c  A1      : 0x3ffd1690  
[21:10:48]A2      : 0x3ffd16c4  A3      : 0xfffffff4  A4      : 0x00000007  A5      : 0x00060023  
[... 96 lines cut ...]
[21:10:54]Subscribing to service notifications of FFC2
[21:10:54]Notifications turned on
[21:10:54]Done with this device.
[21:10:54]BLE connection to Spark established.
[21:10:54]Changing to HW preset 1
[21:10:54]Guru Meditation Error: Core  0 panic'ed (LoadProhibited). Exception was unhandled.
[21:10:54]
[21:10:54]Core  0 register dump:
[21:10:54]PC      : 0x400d310f  PS      : 0x00060e30  A0      : 0x800e5e2c  A1      : 0x3ffd1690  
[21:10:54]A2      : 0x3ffd16c4  A3      : 0xfffffff4  A4      : 0x8008563c  A5      : 0x3ffbee4c  
[21:10:54]A6      : 0x00000000  A7      : 0x003fffff  A8      : 0x00000014  A9      : 0x00000000  
[21:10:54]A10     : 0x3ffbc408  A11     : 0x80000001  A12     : 0x80097d58  A13     : 0x3ffcf270  
[21:10:54]A14     : 0x00000003  A15     : 0x00060023  SAR     : 0x00000020  EXCCAUSE: 0x0000001c  
[21:10:54]EXCVADDR: 0xfffffff4  LBEG    : 0x40090839  LEND    : 0x4009085b  LCOUNT  : 0xffffffff  
[21:10:54]
[21:10:54]
[21:10:54]Backtrace: 0x400d310c:0x3ffd1690 0x400e5e29:0x3ffd16b0 0x400d5f3b:0x3ffd16f0 0x400d6332:0x3ffd1870 0x401e1452:0x3ffd18b0 0x400f7065:0x3ffd18d0 0x400ea0a9:0x3ffd1950 0x400ea0f3:0x3ffd1970 0x400eb50c:0x3ffd1990 0x400e9de5:0x3ffd19f0 0x400e8319:0x3ffd1a10 0x400ef4d3:0x3ffd1a40 0x400edf88:0x3ffd1a70 0x400edf93:0x3ffd1a90 0x400814b6:0x3ffd1ab0 0x400f7bc2:0x3ffd1ad0
[21:10:54]
[21:10:54]
[21:10:54]
[21:10:54]
[21:10:54]ELF file SHA256: 2bc66c482fdf826f
[21:10:54]
[21:10:54]E (3101) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
[21:10:54]Rebooting...

stangreg, 2024-02-18:
@thrivewebdesign I think the interesting point is this one:
[21:10:48]E (12) SPIFFS: mount failed, -10025
[21:10:48]E (13) SPIFFS: mount failed, -10025
[21:10:48]ERROR while trying to open presets list file
[21:10:48]E (24) SPIFFS: mount failed, -10025
[21:10:48]E (25) SPIFFS: mount failed, -10025

Have you loaded the data folder to the Ignitron? Please check [here](https://github.com/stangreg/Ignitron/blob/main/src/Readme.md#installing-firmware-and-data-files) for instructions.

thrivewebdesign, 2024-02-19:
I had previously used the release binary but have built from source today and flashed the firmware again. I can now connect to the spark GO if I turn on the amp then the ignitron then turn the amp off then on again. I can change presets etc ( its fantastic!! ) .  I am still unable to access the presets. Im a bit stumped as to what I'm doing incorrectly as the log from the upload seems ok. 

ESP32 is a WROOM-32

I am using v 1.1 of arduino-esp32fs-plugin

[SPIFFS] data   : /Users/dave/Downloads/Ignitron/data
[SPIFFS] start  : 2162688
[SPIFFS] size   : 1920
[SPIFFS] page   : 256
[SPIFFS] block  : 4096
/HideAndSeek.json
/ParadiseCity.json
/NovemberRainSolo.json
/AcDc.json
/SweetChildOfMine.json
/StairwayToHeavenSolo.json
/WoodenBridge.json
/The1975.json
/Rush.json
/VanHalen.json
/AngryMonkey.json
/VintageCreamyLarson.json
/Oasis.json
/DireStraits.json
/SilverShip.json
/PerfectCleanLarson.json
/EnterSandman.json
/NothingElseMatters.json
/BluesArk.json
/SlashV2.json
/DancingInTheRoom.json
/HateToSay.json
/IronHammer.json
/config/BTMode.config
/CrazyTrain.json
/TremontiClean.json
/MasterOfPuppets.json
/AintTalkinBoutLoveLARSON.json
/FooFightersEverlong.json
/NothingElseMattersMetall.json
/IronMaiden.json
/SultansOfSwingKnopfler.json
/RageNowInTheFire.json
/FuzzyJam.json
/NothingElseMattersAcous.json
/SatrianiSatchBoogie.json
/TornadoOfSouls.json
/FooFighters.json
/Wizo.json
/GunsNRoses.json
/StoneBreaker.json
/HeavyAxe.json
/EddieVanHalen.json
/BritishAccent.json
/RedHotChiliPeppers.json
/LynyrdSkynyrd.json
/JethroTull.json
/Hendrix.json
/BohemianRhapsody.json
/BendTheFend.json
/HouseoftheRisingSonJusti.json
/Megadeth.json
/EricJohnsoncliffsofDover.json
/WizoQuadrat.json
/DookieSoundGreenDay.json
/CountryRoad.json
/ComfortablyNumbSolo.json
/ModernStone.json
/RoyalCrown.json
/PapaResortLastResort.json
/OrangeAD30Arpeggio.json
/DualTrain.json
/MillenialLead.json
/RockDistortion.json
/Anastasia.json
/ToolClean.json
/KnockingOnHeaven.json
/SpookyMelody.json
/SultansOfSwing2.json
/BTMode.config
/CleanForBallads.json
/TremontiAlternative.json
/SlashAFD.json
/PresetList.txt
/ModernVintageClean.json
/SlashDontCry.json
/SunshineOfMyLove.json
/AliceCooper.json
/FinalCountdown.json
/ZakkWyldeNoMoreTearsLead.json
/EnglFireball.json
/SweetMemory.json
/BangBang.json
/12stracoustrummingCN1.json
/DarkSoul.json
/SantanaBlackMagicWoman.json
/GhostDanceMacabreMart.json
/CrazyCrue.json
/VoodooKidWah.json
[SPIFFS] upload : /var/folders/yz/v_0b3m6x7tb74s7k8zck5t880000gn/T/arduino_build_511076/Ignitron.spiffs.bin
[SPIFFS] address: 2162688
[SPIFFS] port   : /dev/cu.usbserial-0001
[SPIFFS] speed  : 921600
[SPIFFS] mode   : dio
[SPIFFS] freq   : 80m

esptool.py v4.5.1
Serial port /dev/cu.usbserial-0001
Connecting.......
Chip is ESP32-D0WDQ6 (revision v1.0)
Features: WiFi, BT, Dual Core, 240MHz, VRef calibration in efuse, Coding Scheme None
Crystal is 40MHz
MAC: xx:xx:xx:xx:xx:xx
Uploading stub...
Running stub...
Stub running...
Changing baud rate to 921600
Changed.
Configuring flash size...
Auto-detected Flash size: 4MB
Flash will be erased from 0x00210000 to 0x003effff...
Compressed 1966080 bytes to 26489...
Writing at 0x00210000... (50 %)
Writing at 0x00232ff5... (100 %)
Wrote 1966080 bytes (26489 compressed) at 0x00210000 in 9.8 seconds (effective 1605.3 kbit/s)...
Hash of data verified.

Leaving...
Hard resetting via RTS pin...

stangreg, 2024-02-19:
Does that mean that you can only switch the HW presets but cannot switch to the custom ones?
Can you please send another log of the Spark when you try to switch between presets?

thrivewebdesign, 2024-02-22:
I have managed to upload the preset data ( used platform io rather than Arduino ) 

I cant seem to establish a pattern for the connection issue, it looks as though the json being broadcast by the Spark go is sometimes not received ( or sent ) as you will see from the log below. 

After the connection is established the ignitrON keeps changing the preset to number 1 ( you can audibly hear this if you drum the guitar ) 

Sometimes it connects first go others it may change to preset 1 - 5 times and connect other times it will just keep changing to preset 1. 

Pressing the preset select button on the spark sometimes breaks this cycle. When this is occurring changing the presets on the ignitron does not seem to work.  

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
Initializing
Operation mode (boot): 1
Last bank not full, filling with last preset to get bank complete
Last bank not full, filling with last preset to get bank complete
Stopping advertising keyboard
Starting scan
Operation mode: 1
======= Entering APP mode =======
Initialization done.
Found Spark, connecting.
Scan ended.
Connected to: xx:xx:xx:xx:xx:xx
Subscribing to service notifications of FFC2
Notifications turned on
Done with this device.
BLE connection to Spark established.
Changing to HW preset 1
Changing to HW preset 1
Changing to HW preset 1
Changing to HW preset 1
Changing to HW preset 1
Changing to HW preset 1
Changing to HW preset 1
Changing to HW preset 1
Changing to HW preset 1
Changing to HW preset 1
Changing to HW preset 1
Changing to HW preset 1
Changing to HW preset 1
Changing to HW preset 1
Changing to HW preset 1
Changing to HW preset 1
Changing to HW preset 1
Changing to HW preset 1
Changing to HW preset 1
Changing to HW preset 1
Changing to HW preset 1
ets Jun  8 2016 00:22:57

----==== PHYSICALLY RESET THE ESP32 HERE ====----

rst:0x1 (POWERON_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
Initializing
[... 96 lines cut ...]
{"Name": "BlueComp", "IsOn": true, "Parameters":[0.5107, 0.4917, 0.4931, 1.0000]}, 
{"Name": "DistortionTS9", "IsOn": true, "Parameters":[1.0000, 0.8106, 0.7192, 0.0000]}, 
{"Name": "6505Plus", "IsOn": true, "Parameters":[1.0000, 0.7624, 0.4000, 0.2793, 0.9083]}, 
{"Name": "ChorusAnalog", "IsOn": true, "Parameters":[0.3104, 0.2797, 0.7458, 0.5678]}, 
{"Name": "DelayRe201", "IsOn": false, "Parameters":[0.2634, 0.2959, 0.7797, 0.4486, 0.0000]}, 
{"Name": "bias.reverb", "IsOn": true, "Parameters":[0.3669, 0.6722, 0.3146, 0.2845, 0.4104, 0.4763, 0.3000, 1.0000]}],
"Filler": "DE"
}
Changing to HW preset 1
Changing to HW preset 2
Message processed:
{"PresetNumber": 0, "UUID": "8BA185FA-4F07-4EB5-841E-6F353C1E3401", 
"Name": "Pantera D1", "Version": "0.7", "Description": "", "Icon": "icon.png", "BPM": 160.0000, 
"Pedals": [
{"Name": "bias.noisegate", "IsOn": true, "Parameters":[0.2412, 1.0000, 1.0000]}, 
{"Name": "BBEOpticalComp", "IsOn": true, "Parameters":[1.0000, 0.5266, 0.0000]}, 
{"Name": "SABdriver", "IsOn": true, "Parameters":[1.0000, 0.7714, 1.0000, 0.0000]}, 
{"Name": "6505Plus", "IsOn": true, "Parameters":[1.0000, 0.3069, 0.0000, 0.3168, 1.0000]}, 
{"Name": "GuitarEQ6", "IsOn": true, "Parameters":[0.9934, 0.5923, 0.5000, 0.5000, 0.3496, 0.3628, 0.7164]}, 
{"Name": "DelayMono", "IsOn": false, "Parameters":[0.3087, 0.4679, 0.7363, 0.6000, 1.0000]}, 
{"Name": "bias.reverb", "IsOn": true, "Parameters":[0.6500, 0.1500, 0.5000, 0.6214, 0.3699, 0.3500, 0.1000, 1.0000]}],
"Filler": "5E"
}
Changing to HW preset 1
Scan initiated

stangreg, 2024-02-22:
I just noticed that you don't have debug logging enabled. Can you please go to Config_Definitions.h and remove the comment before `#ifdef DEBUG`, the rebuild and deploy.
If the Spark has switched successfully to HW preset 1, please also try to switch to a custom preset on a different bank.
Please post the logs here.

thrivewebdesign, 2024-02-23:
Rebooting...
ets Jun  8 2016 00:22:57

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0018,len:4
load:0x3fff001c,len:1044
load:0x40078000,len:10124
load:0x40080400,len:5828
entry 0x400806a8
Initializing
Operation mode (boot): 1
Trying to read file list
Last bank not full, filling with last preset to get bank complete
Last bank not full, filling with last preset to get bank complete
Stopping advertising keyboard
Starting scan
Operation mode: 1
======= Entering APP mode =======
Initialization done.
Found Spark, connecting.
Scan ended.
Connected to: f7:eb:ed:0d:b5:3e
Subscribing to service notifications of FFC2
Notifications turned on
Done with this device.
BLE connection to Spark established.
Initial boot, setting preset to HW 1
Changing to HW preset 1
Clearing response buffer.
Sending block:
01FE000053FE1A000000000000000000F00101000138000000F7
Done with this command!
Getting current preset from Spark
Clearing response buffer.
Sending block:
01FE000053FE3C000000000000000000F0010101020100010000000000000000000000000000000000000000000000000000000000000000000000F7
Done with this command!
Received message via BT callback
Received data:
F00101000438F7
Received block with no header start, appending to previous (if present)
Acknowledgement for command 38
Received message via BT callback
Received data:
F001010703012012001901005924004237434346
Received block with no header start, appending to previous (if present)
Block not ready for processing, skipping further processing.
Received message via BT callback
Received data:
453700362D443345342D00344137382D4234F7F0
Received block with no header start, appending to previous (if present)
Block not ready for processing, skipping further processing.
Received message via BT callback
Received data:
01010A03010012011938342D4500443837364645
Received block with no header start, appending to previous (if present)
Block not ready for processing, skipping further processing.
Received message via BT callback
Received data:
3310323542452F5374007564696F207472F7F001
Received block with no header start, appending to previous (if present)
Block not ready for processing, skipping further processing.
Received message via BT callback
Received data:
013C0301001202196970
Received block with no header start, appending to previous (if present)
Block not ready for processing, skipping further processing.
[... 921 lines cut ...]
Done with this command!
Initial boot, setting preset to HW 1
Changing to HW preset 1
Clearing response buffer.
Sending block:
01FE000053FE1A000000000000000000F00101000138000000F7
Done with this command!
Getting current preset from Spark
Clearing response buffer.
Sending block:
01FE000053FE3C000000000000000000F0010101020100010000000000000000000000000000000000000000000000000000000000000000000000F7
Done with this command!
Initial boot, setting preset to HW 1
Changing to HW preset 1
Clearing response buffer.
Sending block:
01FE000053FE1A000000000000000000F00101000138000000F7
Done with this command!
Getting current preset from Spark
Clearing response buffer.
Sending block:
01FE000053FE3C000000000000000000F0010101020100010000000000000000000000000000000000000000000000000000000000000000000000F7
Done with this command!
---=== Turned off spark GO ===---
Scan initiated

stangreg, 2024-02-23:
Thanks for the logs. I have not seen this issue in this form before. What I have noticed is that communication with the Spark 40 is much more stable than with the GO, which could have to do with the difference in message sending I have seen. I think to increase stability on the BT connection, I would need some restructuring of the code. Until I have time to work on this, you could try to check out and build the branch `20240214_refactor_msg_processing`, where I did some changes to improve stability around the GO communication. Note that this branch is in development/testing status. Please let me know if that works better for you.

stangreg, 2024-02-25:
@thrivewebdesign I have further worked on the code in the branch `20240214_refactor_msg_processing`.
Messaging to the GO should be much more stable. However, some presets don't get properly updated, I am yet to find the reason. Can you please test by checking out the branch, building and deploying? This could also help me in further investigations on the stability.

stangreg, 2024-02-25:
@thrivewebdesign I have finally finished the refactoring on message processing and merged that back into the main branch. If you want, you can test it now. Please let me know if it works so we can close this issue

thrivewebdesign, 2024-02-26:
I will download build and test the evening and report back. Thank you so much for hard work.

stangreg, 2024-02-26:
@thrivewebdesign I did a few performance tweaks and code cleanup to make the code better readable. If you update your code to the latest from the main branch, you should have the final version for now.

stangreg, 2024-02-28:
@thrivewebdesign I fixed another bug, I have also built in the automatic switch between Spark 40 and Mini/GO messages (with/without header, message length). This should be it, I have tested it with my Spark 40 and the GO.

thrivewebdesign, 2024-02-28:
Thats wonderful, I managed to kill my only 38 pin ESP32 so currently hacking together a test board and will report back. Thanks again for all your hard work, very much appreciated.

stangreg, 2024-02-28:
Oh no, that is a pity! Hope you can find some replacement. Once you manage to test, please let me know if the issue is resolved for you.

stangreg, 2024-03-12:
@thrivewebdesign FYI, I did some more improvements to the code which should hopefully make the connection to Spark GO more stable. Did you have a chance yet to test?

thrivewebdesign, 2024-03-19:
Apologies for the delay in getting back to you, I've hobbled together a test rig on strip board and a smaller esp32 I had and it is working so much better thank you. The only issue I noticed was when imitating the tuner when holding the preset switching button ( not an issue for me )  Ill test the latest version and report back. FIY PCB's have arrived and i've started assembling in readiness for the new 38pin esp's arrival. Thanks again for creating such a fantastic project.

stangreg, 2024-03-20:
@thrivewebdesign thanks for testing and the kind words!
I am glad that it works for you now.
One question before I close this ticket, what is the issue when you hold the preset button? I just tested it and it changes to the Go tuner mode. When pressing again, it switches back to the preset it had before. Can you please provide more information on the behavior in your case?