stangreg/Ignitron #38: [1.7.3][Boot failed] SPIFFS error

bingliang-zh, 2025-01-28:
## Description

ESP32 boot failed, blank screen.

Log:

```
--- Terminal on /dev/cu.usbserial-0001 | 115200 8-N-1
--- Available filters and text transformations: colorize, debug, default, direct, esp32_exception_decoder, hexlify, log2file, nocontrol, printable, send_on_enter, time
--- More details at https://bit.ly/pio-monitor-filters
--- Quit: Ctrl+C | Menu: Ctrl+T | Help: Ctrl+T followed by Ctrl+H
E (664) SPIFFS: mount failed, -10025

assert failed: xTaskGetIdleTaskHandleForCPU tasks.c:3045 (( xIdleTaskHandle[cpuid] != ((void *)0) ))

Backtrace: 0x40083e9d:0x3ffe3990 0x4009508d:0x3ffe39b0 0x4009ae71:0x3ffe39d0 0x40096b7e:0x3ffe3b00 0x40105e6c:0x3ffe3b20 0x400ee79b:0x3ffe3b40 0x400ee809:0x3ffe3b60 0x400e2111:0x3ffe3b90 0x400e915e:0x3ffe3bb0 0x400e91e3:0x3ffe3be0 0x400d45f7:0x3ffe3c00 0x40112ef3:0x3ffe3c20 0x400835b1:0x3ffe3c50 0x40079306:0x3ffe3c90 |<-CORRUPTED

ELF file SHA256: b1f57130a9ad4522

E (699) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
E (664) SPIFFS: mount failed, -10025

assert failed: xTaskGetIdleTaskHandleForCPU tasks.c:3045 (( xIdleTaskHandle[cpuid] != ((void *)0) ))

Backtrace: 0x40083e9d:0x3ffe3990 0x4009508d:0x3ffe39b0 0x4009ae71:0x3ffe39d0 0x40096b7e:0x3ffe3b00 0x40105e6c:0x3ffe3b20 0x400ee79b:0x3ffe3b40 0x400ee809:0x3ffe3b60 0x400e2111:0x3ffe3b90 0x400e915e:0x3ffe3bb0 0x400e91e3:0x3ffe3be0 0x400d45f7:0x3ffe3c00 0x40112ef3:0x3ffe3c20 0x400835b1:0x3ffe3c50 0x40079306:0x3ffe3c90 |<-CORRUPTED

ELF file SHA256: b1f57130a9ad4522

E (699) esp_core_dump_flash: Core dump flash config is corrupted! CRC=0x7bd5c66f instead of 0x0
Rebooting...
ets Jul 29 2019 12:21:46

rst:0xc (SW_CPU_RESET),boot:0x13 (SPI_FAST_FLASH_BOOT)
configsip: 0, SPIWP:0xee
clk_drv:0x00,q_drv:0x00,d_drv:0x00,cs0_drv:0x00,hd_drv:0x00,wp_drv:0x00
mode:DIO, clock div:2
load:0x3fff0030,len:1184
load:0x40078000,len:13232
load:0x40080400,len:3028
entry 0x400805e4
E (664) SPIFFS: mount failed, -10025

assert failed: xTaskGetIdleTaskHandleForCPU tasks.c:3045 (( xIdleTaskHandle[cpuid] != ((void *)0) ))

Backtrace: 0x40083e9d:0x3ffe3990 0x4009508d:0x3ffe39b0 0x4009ae71:0x3ffe39d0 0x40096b7e:0x3ffe3b00 0x40105e6c:0x3ffe3b20 0x400ee79b:0x3ffe3b40 0x400ee809:0x3ffe3b60 0x400e211^C^C
Aborted!

 *  The terminal process "platformio 'device', 'monitor', '--port', '/dev/cu.usbserial-0001'" terminated with exit code: 1. 
 *  Terminal will be reused by tasks, press any key to close it. 

```

## Mitigate method

Remove `SPIFFS.begin(true);` in `SparkPresetBuilder.cpp` will let ESP32 to boot into APP mode. This change is introduced in this recent [commit](https://github.com/stangreg/Ignitron/commit/961f259ab04f0acefe08baa1147baca83f446daf). But haven't test it in other modes.

## Environment

I'm using a 38 pin ESP32 dev board with CP2102 and typec connector. Chip is ESP32-WROOM-32U.

stangreg, 2025-01-28:
@bingliang-zh I am not 100% sure if the error is caused by the SPIFFS.begin(true) line, but I have reverted it for now as it was working for me before and after the change. I only introduced this line as it was recommended by another user who had problems before. I will keep the line out for now until I have further knowledge on any potential issue and solution.

barsk, 2025-01-30:
This is strange. I and another user had initial problems with SPIFFS not being formatted on a brand new ESP32 (38 pin). Adding SPIFFS.begin(true) fixes that as it reformats the partitions according to the partition data. It only needs to be run once though, after that the partitions is in sync with the code and the hardware, and SPIFFS.begin(true) will have no ill effect.
The error we got initially without the SPIFFS.begin(true) was:
`SPIFFS: mount failed, -10025`

I have no idea why this caues a problem for @bingliang-zh. Maybe that actual actual ESP32 is of another model that would need an alternate configuration setup in platformio.ini?

I'm no expert in ESP32 or VS Studio PlatformIO, this is just my thoughts.

stangreg, 2025-01-30:
When building and deploying the firmware, the foreseen activities include the upload of the data folder to Ignitron. As far as I understand, this includes the formatting of the file system. Without the upload, Ignitron will not work properly and there would be no custom presets saved on Ignitron. Currently, there is no SPIFFS.begin() at all in the code and it works fine with my ESP32. 
@barsk when you faced this issue, did you upload the data folder to Ignitron before?

barsk, 2025-01-30:
I built the firmware and uploaded (installed) on the ESP32. Just to try out the build and firmware worked. No data folder upload. Something I missed?

stangreg, 2025-01-30:
@barsk yes you are missing uploading of the presets which are in the "data" folder.

This is from the Readme file:

"After building and installing the firmware on the board, it is required to also transfer the data directory to the board. In order to do so, use the PlatformIO targets 'Build Filesystem Image' and 'Upload Filesystem image'. You might need to specify the correct partitioning in the platformio.ini file. Please make sure to set the board settings correctly before uploading (see above).
"

bingliang-zh, 2025-01-30:
I found the change by AB testing different commits. Once I removed it, my ESP32 works. Just to confirm it leads to my issue, I don't know what is the root cause.

@barsk This is my device:
[image]

barsk, 2025-01-30:
Ok. Missed that! Sorry.
So uploading would then fill the SPIFFS file systems properly and mitigate the the Boot failed problem?

barsk, 2025-01-30:
Ok, I have just received the PCBs from Aisler. They look good and for once I chose the easy path instead of working out my own solution. So, going to wire it all up and install in a 3D-printed case. Will take some time, but I'll come back here with how it all works with everything set up correctly and if the error persists or not.

stangreg, 2025-01-30:
> Ok. Missed that! Sorry. So uploading would then fill the SPIFFS file systems properly and mitigate the the Boot failed problem?

@barsk Yes, I assume so.

barsk, 2025-02-03:
I have now built the latest firmware and  transferred the data directory to the board properly. There are no problems reported on the serial terminal and everything seems to work perfectly. 
Initializing
Operation mode (boot): 1
Initializing PresetBuilder
**Reading custom presets from filesystem.**
Reading opmode file.
OPMode: 1
Reading operation mode from file: 1
Stopping advertising keyboard
Starting scan
Operation mode: 1

I would assume SPIFFS.begin(true) is not necessary when the filesystem image is properly uploaded.

stangreg, 2025-02-04:
Ticket closed because FS has been migrated from eSPIFFS to LittleFS.

closed 2025-02-04 by stangreg