stangreg/Ignitron #52: Display is inverted when returning from Tuner mode

barsk, 2025-03-19:
**Describe the bug**
When in Tuner mode and you have the string perfectly tuned, that is indicated by the display turning inverted. If you return to normal mode the display will be inverted there too which it is in error. 
This bug seems to be only present with #define DEDICATED_PRESET_LEDS in Config_Definitions.h and may be related to the PRESET MODE label which is printed inverted at the bottom line in the display. Somehow the invertion of the display gets out of "sync".

**To Reproduce**
Steps to reproduce the behavior:
1. #define DEDICATED_PRESET_LEDS in Config_Definitions.h
2. Install firmware
3. Start Ignitron and Spark amp so it is connected
4. Make sure PRESET MODE is displayed
5. Long press on "Noise" button enters Tuner mode
6. Tune string to perfect and the display will invert
7. While inverted return to normal (PRESET MODE)
8. The display is now inverted from what it should be

**Expected behavior**
The display in PRESET MODE should be displayed the same regardless if Tuner mode was used before or not.

stangreg, 2025-03-19:
@barsk I have implemented a fix in branch **20250319_issue_52_bf**. Maybe you can test it with your additional LED setup and let me know if it works. I have changed the general logic how to invert the display, this should be more consistent.

barsk, 2025-03-20:
@stangreg Seems to work fine!
When you have merged that branch I will prepare a pull request with my changes.

stangreg, 2025-03-20:
@barsk thanks for testing. With the fix in place, do you think your changes are still required? Have in mind that key information should be readable in the display even when standing, so maybe font size 1 is a bit too small. I personally like how your first implementation looks.

barsk, 2025-03-20:
@stangreg I did not do any fixes besides removing my inverted settings, and that seemd to solve the issue. 
I did however see the problem when the amp goes disconnected, and the display was in inverted mode when it disconnected. Then the IGNITRON logo (idle mode) displays inverted too. That seems to have been fixed now too.

The smaller font may be an issue, although the bigger 1.3" display counters that a bit. I confess that I use the Ignitron mostly sitting down with the pedal next to me on a desktop. In that use case the smaller font looks pretty nice indeed. Here is a photo of it in use:

[image]

Maybe I can make the changes configurable? It does not affect much code at all. Then we can have it both ways. 👍 

The other change is that I made the FX Leds blink while in Manual/FX mode. That makes it so much clearer that we are not in a normal mode and the FX buttons are what we can push now. I used mostly the same logic as you did in your blink code. Looks terriffic IMHO. :smile:

stangreg, 2025-03-21:
@barsk I would like to keep the code as simple as possible in order to keep maintenance simple and to keep the code small. So I think it is not required to make this cosmetic part configurable.
On the blinking LEDs, I am not sure if that might be a bit irritating if the whole pedal would be blinking while playing. Of course you can adjust your code as you like, but I am not sure if I would take it into the main branch of that was your plan.

closed 2025-03-22 by stangreg

barsk, 2025-03-22:
Well, my usage scenario for the manual/fx mode is to make a quick change to the current mode, and then revert back to normal preset mode. In that case having the LEDs blink makes it super clear that you are in a temporary mode which I think helps.

However, if people are using the manual mode regularly, and on stage, then the blink mode would of course be irritating. I'll put this change in a branch of my own fork. Then you could check it out and see if you like it or not.

Btw, I do understand that you are the owner of this project and that you do make the decisions, and I have no problem with that whatsoever. I make these changes as a suggestion. If it is turned down and I absolutely want it, I will have these modifications in my own fork. 

Btw again, having configuration settings as defines makes the code rather ugly with a lot of #ifdefs. Maybe you could consider having regular variables based on enums for some of this. That would make it easiser to incorporate small variants, and it would also open up for using object orientation methods like factory patterns to instanciate the concrete variant classes that do the LED and display renderings. Although that would of course make the code a bit larger and complex. Again, just a thought. Probably not worth it... 😎

stangreg, 2025-03-22:
@barsk I really appreciate your input and I think there have been some very useful adjustments, e.g. the additional optional LEDs. I am always open for suggestions while trying to keep the code clean and not too convoluted.
We seem to have similar ideas, as I have just merged a branch into main where I have replaced a lot of #defines with enum types or constants. I think this makes the code already much cleaner and also more stable as the types are now enforced.
Maybe you want to test the latest main branch if it still works fine with your additional LEDs (otherwise I might have done something wrong while moving to enums).

I have previously also thought of implementing a kind of factory approach (mainly to separate APP and AMP mode more), however, I have then decided that this would require a LOT of refactoring, so I left it there.

barsk, 2025-03-24:
Good to hear! I will check out the latest main and have a look.
I may drop the smaller font idea to keep the code in sync with yours, and that leaves the blink mode of the FX/MANUAL as a case I really think is useful though. I will have alook at if and how it can be incorporated with the least amount of code changes. 

Keeping the code small and simple is a great goal. Simple is beautiful.

barsk, 2025-03-24:
I found no problems with the latest 1.8.6 version. Works fine in both DEDICATED_PRESET_LEDS mode and not.

There is a bug with Tuner LEDs though, but it is a older one that has been there for some revisions. I have created an issue #53 for that. I also made an PR #54 regarding that (and the other stuff).